moeiscool/Shinobi #1: First run problems

muldy, 2016-12-15:
Hi!
I'm just taking this for a spin on raspberry pi:

I had  to change database name to other name, on the framework.sql (commented the create database line)
I had to change to port 8065, since 80 is already taken :)

got this on start:
uncaughtException { [Error: spawn ENOENT] code: 'ENOENT', errno: 'ENOENT', syscall: 'spawn' }
{ [Error: Cannot enqueue Query after fatal error.] code: 'PROTOCOL_ENQUEUE_AFTER_FATAL_ERROR', fatal: false }

And I can't login...

muldy, 2016-12-15:
changed conf.json too

moeiscool, 2016-12-15:
The error should now be "resolved" in the newest commit. It means you don't have dstat installed. I added a catch so it shouldn't get in the way.

 I forgot to mention that in the readme previously. The readme now includes this just after the `Application Install` section.

can you verify in your database you have the default user? `ccio@m03.ca` password is in an MD5 string `password`.

You may need to select the Users table and insert `default_data.sql` from the sql folder.

If not you can insert your own user.

with this SQL command. This will act as the `default_data.sql`
```
INSERT INTO `Users` (`ke`, `uid`, `auth`, `mail`, `pass`, `details`) VALUES ('2Df5hBE', 'XDf5hB3', 'b1f4fc2d0f8b2707fef62e0fd47e3d57', 'ccio@m03.ca', '5f4dcc3b5aa765d61d8327deb882cf99', '{"days":"0.5"}');
```

moeiscool, 2016-12-15:
for a default RTSP monitor row this SQL row can be used.
```
INSERT INTO `Monitors` (`mid`, `ke`, `name`, `shto`, `shfr`, `details`, `type`, `ext`, `protocol`, `host`, `path`, `port`, `fps`, `mode`, `width`, `height`) VALUES ('bunny', '2Df5hBE', 'Bunny', '[]', '[]', '{"model":"RTSP Stream"}', 'h264', 'mp4', 'rtsp', '50.73.56.89', '/axis-media/media.3gp', 80, 1, 'record', 640, 480);
```

muldy, 2016-12-16:
hmm

Looks good:

`mysql> select MD5("password");
+----------------------------------+
| MD5("password")                  |
+----------------------------------+
| 5f4dcc3b5aa765d61d8327deb882cf99 |
+----------------------------------+
1 row in set (0.00 sec)

mysql> select * from Users \G;
*************************** 1. row ***************************
     ke: 2Df5hBE
    uid: XDf5hB3
   auth: ec49f05c1ddc7d818c61b3343c98cbc6
   mail: ccio@m03.ca
   pass: 5f4dcc3b5aa765d61d8327deb882cf99
details: {"days":"0.5"}
1 row in set (0.00 sec)`

moeiscool, 2016-12-16:
Hmmm. Does the node app throw any errors in regards to SQL? does the node app even start? Did you install dstat? You don't need to though I tried it yesterday without and it worked for me.

Is sql on the same machine you are trying to install Shinobi on? Only thing I can think is that the node app is not authenticating with your sql server.

moeiscool, 2016-12-16:
in the latest commit you can now register accounts from the login page. Please try that, if it doesn't create an account i suspect your app is not connecting to the database.

muldy, 2016-12-16:
Strange 

`Error Connecting : DB { [Error: ER_ACCESS_DENIED_ERROR: Access denied for user 'shinobi'@'piserver.lan' (using password: YES)]`

should be shinobi@127.0.0.1 
as in config:
{"host":"127.0.0.1","user":"shinobi","password":"passhere","database":"ccio"}

User shinobi has no access from remote

moeiscool, 2016-12-16:
hmmm i have never seen that happen. is `piserver.lan` your hostname? is this a fresh install?
do you have any programs installed that might conflict?

You can allow `piserver.lan` as a host but.. i don't think that's a good idea unless the username and pass are something crazy unguessable. Even then it might be considered a serious security hole :/

muldy, 2016-12-16:
ok, allowing from remote worked, even though on config is 127.0.0.1 or localhost.. (I'm aware of the security implications)
Yes, piserver.lan is my hostname.
Gonna check the app :) Thanks!

moeiscool, 2016-12-16:
sweet :) gonna close this issue. If you have anymore trouble you know where to find me! :) happy trails!

closed 2016-12-16 by moeiscool