moeiscool/Shinobi #3: Camera Implementation

Whytehorse, 2016-12-23:
Which cameras was this implemented for? Would it work for a raspberry pi streaming RTSP h264 video? Does it record continuously? Does it use motion detection? If so how does it do that on h264? Does it timestamp the videos? Is the video montage showing the actual stream or a reduced version? Imagine 15 1080p IP cameras, the montage can easily overload the NIC on the server...

antares2001, 2016-12-23:
Hi @Whytehorse 

I'd first thank the author for his great effort in building a new and better open source system for camera surveillance than what is available today.

Yet, I have the same questions and I think, that it would be good to address them in the project wiki in a FAQ section.

I am going to install this now and see how it works. I might report back my findings.

Again: thanks for your great effort and keep the work going. Though I cannot help to program, I hope you will find people to build a team that will create great software (yet, I'm not convinced that Node.js is the right thing to do this -- but that is something that might end in programming language flame wars ;-)).

moeiscool, 2016-12-23:
**Which cameras was this implemented for?**
When i started MJPEG, and JPEG.
It does : MJPEG, JPEG and RTSP
I am looking into adding local cameras, but later.

**Would it work for a raspberry pi streaming RTSP h264 video?**
If you can open the stream in VLC Media Player then ffmpeg should be able to get it too. ffmpeg is what Shinobi uses for encoding.

**Does it record continuously?**
yes, does a cutoff at 15 minutes (hardcoded for now). 

**Does it use motion detection?**
I am going to be implementing motion detection soon, just need to make it use less overhead.

**Does it timestamp the videos?**
It does but for some wacky reason the time is off sometimes. I am trying to figure out why it does that. I am welcoming any input on this.

**Is the video montage showing the actual stream or a reduced version?**
I set it to reduce the size to 640x480 i think because when I had 2048x1536 frames some of them didn't even load because it would take so long for the browser to render them and before it could render it would get replaced.. essentially you are getting the stream but you just cant see anything.

**Imagine 15 1080p IP cameras, the montage can easily overload the NIC on the server...**
Word. I actually have about 14 cameras myself. I had 10 of them running in the demo images you see in the readme on a nooby little macbook. Seemed to be okay to me, but I am on the same network. When i open up the Task Manager on windows to see my NIC usage the spikes are pretty low... again probably because I on the same network. Well the program is in infancy and I stress testing is going need to happen. :)

much appreciated! :) as you can probably tell from my username I enjoy praise! 

**Though I cannot help to program, I hope you will find people to build a team that will create great software (yet, I'm not convinced that Node.js is the right thing to do this -- but that is something that might end in programming language flame wars ;-)).**

LOL that just made my day.

closed 2016-12-23 by moeiscool

Whytehorse, 2016-12-23:
Yes I think node.js makes perfect sense for modern surveillance systems. I can help with coding and testing but don't know these particular systems. 

As for my application, I am using vlc on the rpi to serve up h264 rtsp streams and on my server I use avconv(ffmpeg) to dump the video to disk and serve up another stream for a live view.  Here is the code:
Record from an IP camera:
`avconv -i rtsp://192.168.1.XXX:8554/ \ 
-acodec copy -vcodec copy -map 0 \ 
-f segment -segment_time 3600 -segment_format mp4  \ 
-crf 22.0 "/events/video/NEcornerW/`date +%Y-%m-%d-%H-%M-%S`-capture-%03d.mp4" &`

Stream from camera:
`LD_PRELOAD='' cvlc v4l2:///dev/video0 --v4l2-width 1280 --v4l2-height 720 --v4l2-chroma h264 --sout '#rtp{sdp=rtsp://:8554/}' `

Whytehorse, 2016-12-23:
As for the timestamp, there is no real-time clock on the rpi so the clock drifts until it can be corrected by ntp. My server has real-time clock and the timestamps are perfect but the video from the rpi is stretchy and has a lag due to buffering in vlc and the network. It's about 500msec behind.

scottlamb, 2016-12-29:
Happened upon your project and saw your note about timestamps here:

> Does it timestamp the videos?
> It does but for some wacky reason the time is off sometimes. I am trying to figure out why it does that. I am welcoming any input on this.

I'm also working on an NVR project and have thinking about this exact problem recently. I'm writing a doc about it; the current version is here: https://github.com/scottlamb/moonfire-nvr/blob/063708c9abb222c65051575a797e6e6c670dbaa0/design/time.md