moeiscool/Shinobi #4: Installation without root privileges

antares2001, 2016-12-23:
Hello!

Thanks a lot for the great work! A question that has come to me during the performance of a test install of Shinobi is: could you provide an installation instruction, that does not require root privileges for Shinobi? When looking at the purpose of the program, I think that it does not need root privileges and thus it's much more secure to run it as "normal" user than as root.

Or is there a reason for elevated privileges?

moeiscool, 2016-12-23:
agreed. @curioussavage suggested this and I will be changing the default port to 8181

moeiscool, 2016-12-23:
port default will be 8080, not 8181

closed 2016-12-23 by moeiscool

antares2001, 2017-02-09:
Hey @moeiscool

It's really great to see that the project is making progress. With a job and family, it's unfortunately sometimes hard to follow regularly.

Has the issue with root been solved, i.e., does Shinobi run without root privileges? I think I will give it a try in the next days again, to see whether I can remove my Zoneminder installation in favor of Shinobi.

moeiscool, 2017-02-09:
Yes Shinobi can be installed and ran as a regular user. Sweet, I'd say wait for a few days. The dashboard will be mobile ready soon.

antares2001, 2017-02-13:
Great! Thanks for the update. Are you also planning on an openHAB binding for Shinobi?

moeiscool, 2017-02-13:
I have no plans for openhab in specific but I know @rockedge uses openhab to send mode changes to his cameras over a http request. In my short reading about openhab it seems this is the preferred method of using openhab for users, rather than integrating features specifically for openhab to just use http URLs it can connect to. https://github.com/moeiscool/Shinobi/wiki/API-Access#set-mode-of-camera-record-watch-or-stop-from-http

antares2001, 2017-03-21:
Hi @moeiscool and thanks for the update.

Sorry, wasn't able for some time to get back to the machine. Have now a fresh Debian 8 installation and will give Shinobi a try tonight. You're right about openHAB. I had it working with Zoneminder (as provider for the video stream) and sent alarm events via the api so it recorded when my PIR or window sensor detected a motion / open window.

Does Shinobi provide a method to trigger saving the camera stream from the outside via http request?

Thanks again for your great work. Also going to donate for it when I get it working! I'm sure it's worth it as the project seems not to die.

antares2001, 2017-03-21:
Hmmm... unfortunately installation still won't work out of the box as non-root user. As I don't have time to investigate more (but I just saw it doesn't start up properly and throws exceptions in the logs), I'll come back at a later stage to try it again. I think you understand, I'm not a friend of running stuff in root mode, when it's not necessary to do so.

Anyhow keep up with the great work, that I'll hopefully be able to test one day.