moeiscool/Shinobi #7: Debugging Camera flows

mrproper, 2016-12-24:
Hey, thanks for the work so far, finally an alternative to zm starting to brew yay.
Im trying to get an rtsp stream running, i can run the stream in vlc with no problems, but when i replicate the same thing in shinobi it gives me no image error.

After some poking around it looks like the ip camera sends back a tcp reset.

I cannot find how shinobi is connecting to the camera with ffmpeg to replicate it just with ffmpeg.

Also since it uses ffmpeg it should support basic auth rtsp, but theres no fields to put this into shinobi

Cheers!

moeiscool, 2016-12-24:
Sorry about that, right now i am unable to get a snapshot from RTSP or MJPEG. I honestly have no clue why it doesn't work but it doesn't so i commented it out and made it default to "No Image". For now you need to click the monitor to see if its actually running or not.

usernames and passwords can be put into the camera URL like this
username:password@111.111.111.111

There have been a handful of times I thought my cameras weren't working but I just ended up putting the wrong port number :rofl: please make sure the port is in the port field and not trailing the host.

You can also open your browser console to see when you get a log.
In the next commit there will be a more human way to look at the incoming logs.

what model is the camera?

moeiscool, 2016-12-24:
I just did an update to camera.js please try again.

mrproper, 2016-12-24:
I added a console statement to the camera.js:
                                 if(e.type==='mjpeg'||e.type==='h264'||x==='record'){
+                                    console.log('DEBUG! ffmpeg');
+                                    console.log(s.ffmpeg('args',e));

just to see how it was getting the stream, I worked out my issue from that, my camera is some HikVision dome camera.

I like the work so far, though im digging into why the mp4 storage wouldnt work and why webm just killed all of my ram then ate all my swap. It could be that i was doing 28fps with 3k resolution capture =)

moeiscool, 2016-12-24:
i experienced serious hanging on my macbook last night i was so tired i just went to sleep. That's probably the version you have, please redownload. I noobed out and committed before testing.

even on 3k, if you have only one going it shouldn't each much power.. but 28 fps might do it LOL.. most of my cameras I only do 10 max.. even then its usually just 5. Looks smooth :D Long Live FFMPEG!

mrproper, 2016-12-24:
With all the hidef cameras these days. What I'm interested is just
recording multiple cameras all the time. Then I just want scene changes to
tag events in the stream. Then another process to just expire non event
tagged streams after a set period. So the front end preview should just be
another stream at a lower fps/resolution most new cameras offer multiple
rtsp streams at different res and fps. Ill do further testing and get back
to you

<https://github.com/moeiscool/Shinobi/issues/7#issuecomment-269101299>, or mute
the thread

moeiscool, 2016-12-24:
#3 checkout this issue. Some good questions were asked there and might give you a better picture of what shinobi is doing. you can also ask more questions there if you want ill answer in the same format.

i made forum on freeforums but honestly i don't even go there.

mrproper, 2016-12-24:
I had a look at that. Should probably dump a lot of those questions into
the github wiki

moeiscool, 2016-12-25:
Hey did you try running Shinobi in non root user? Some cameras don't show for me and just give error and stop. Was wondering if you had any input on this. Why is ffmpeg unable to spawn unless pm2 start cameras.js is done in root :/

moeiscool, 2016-12-25:
lolcats the problem was user rights on the events folder. one of those days.

closed 2016-12-25 by moeiscool