moeiscool/Shinobi #9: Streamer v1 memory leak on intake, anyone got any ideas?

moeiscool, 2016-12-27:
when the frames are being input from the client the memory just gets eaten. Does anyone know of a way to clear this memory after its been dealt with in said emit? or just clear it in general?

websocket is done through socket.io btw.. so its an issue with socket.io but its probably in how im using it. 

To reproduce this you grab the newest commit of Shinobi, it has Streamer v1 but you'll need to update your db structure to use it of you have an old version going.

If you don't want to overwrite your current install (which is probably for the best) you can grab the virtualbox VM and update that to check it out.

you can see the memory usage of the pm2 process by running
`pm2 list`

kevinGodell, 2016-12-27:
Are you still running the server on you mac? How many cameras are you using? Are you saying a bug was introduced recently when you changed "Streamer" or was it present from the beginning? I set up an old pc with a fresh install of ubuntu 16lts and seem to have no problem with memory leak, but I am using a shinobi version from a couple days ago. Currently, I have 2 rtsp h264 ip cameras set at 704x480 outputting a frame rate of 15fps, with both set to "watch." I just ran "pm2 list" and see I have an uptime of 32 hours with about 53MB of memory.  I will set both to "record" and check the memory later today and report back.

moeiscool, 2016-12-27:
53mb sounds right. Your installation is fine in this regard. The issue is with the Streamer function. If you've tried it you'll notice the memory keep climbing but never go down. It happens when the frames come in through web socket.

If I have no way to fix this, I'll make it builds video on client side instead and post every few seconds (the way facebook does it) my only issue then becomes the video file on the server.

If I am posting webm segments, how will I input them into the event? I don't want to have to stitch them as they come in because ffmpeg won't  be able to keep up when the event gets big enough.

If there was som thing like image2pipe but for video files, that would solute all this in an instant.

moeiscool, 2016-12-28:
not sure what i did but the newest commit doesn't have the Streamer problem. Can you confirm if you are experiencing the issue?

i think what will do is make two version of streamer to use. one is this audio-less version with frames emitted like a snapshot stream.

and another (v2) will be videos created on the client side and sent over every few seconds. hopefully this will address the audio issue. :D

moeiscool, 2016-12-28:
ok nevermind it does still happen.
it happens when the streamer client is firefox on windows
it does **not** happen on android chrome

i did intend this to be used on mobile so i guess its not a serious issue at this time.

closed 2016-12-28 by moeiscool