moeiscool/Shinobi #10: Camera Controls

kzap, 2016-12-29:
would this have any support for camera controls or those are only native to the webserver/ mobile app that came with camera?

i have one that has a http interface which controls the camera and then one uses a mobile app to control the camera... 

isnt there some standard to PTZ camera controls for all these china made devices?

moeiscool, 2016-12-29:
i have been looking into adding camera controls it doesn't seem difficult for foscams, they also uses a series of different HTTP. (they are the only PTZ i have to test with at this time)... but if you give me your camera model i can try and make some code for it and the foscam... to sort of start a universal method of controlling cameras.

moeiscool, 2016-12-31:
newest commit now has control options. please check it out.

it works with my foscam.

moeiscool, 2016-12-31:
here is a video of using foscam control URLs in Shinobi. You can skip half way to see the input settings.
https://www.youtube.com/watch?v=zjs-8LZDc1I

kzap, 2016-12-31:
Thanks,i have one that has http controls. Ill see if i can implement

https://www.youtube.com/watch?v=zjs-8LZDc1I

or mute
the thread

moeiscool, 2016-12-31:
hey just noticed a bug in the code, please download the newest commit or the camera control wont work.

closed 2016-12-31 by moeiscool