moeiscool/Shinobi #11: Timestamp on recorded video

kevinGodell, 2016-12-29:
Is it possible to have the timestamp on the recorded video to be an option? My ip cams already have a title and timestamp on screen from the source, so having 2 timestamps seems redundant. Also, it seems to increase the cpu load of ffmpeg when I monitor it in htop.

kevinGodell, 2016-12-29:
> Sure, ill add an option in the next commit.
> Eh? htop makes ffmpeg use more CPU load? I honestly don't know why that would happen. Maybe it's reading something from each process and ffmpeg can't handle it while it's encoding? I'm just guessing. I don't know how htop works.

Htop is a nicer version of top. It is used to monitor resources such as CPU and ram. I was using htop to monitor CPU usage and noticed that ffmpeg used a bit more when I deployed your latest commit to my Ubuntu server. Adding any extra tasks to ffmpeg will make it use more CPU, that's why I ask for the option to remove the time stamp. Thanks.

moeiscool, 2016-12-29:
Hmmm, it uses more after the last commit? Perhaps your fps isn't set? I made some modifications to how *Record FPS* works. Irma the field is blank it will neglect using the frame rate option instead of default to 1.

moeiscool, 2016-12-29:
i just noticed "watch only" causes the stress on the cpu load.. fixing

moeiscool, 2016-12-29:
Release *Sunlight* has option to remove timestamps. The box will appear near the end of the modal when record is selected.

closed 2016-12-29 by moeiscool

kevinGodell, 2016-12-30:
nice update. I just ran various tests for the last couple hours with various configurations. cpu seems very good now. 1 little bug is present. i will open a new issue for that.