moeiscool/Shinobi #13: ffmpeg crashes if wrong combination of settings

kevinGodell, 2016-12-30:
On my current setup, I have a rtsp camera set to record, mp4 file type, record speed 1, stream video fps=1. If timestamp is "yes" and video codec is "copy," then ffmpeg attempts to start and crashes(monitoring via htop). If timestamp is "yes" and video codec is libx264 or libx265, everything works ok. Also, if timestamp is "no,"  all 3 video codec options work. It only fails when timestamp is yes and video codec is copy.

moeiscool, 2016-12-30:
interesting. i will run some tests to see what can be done about it. it may just be syntax for the ffmpeg command... just one extra space will crash the process lol...

for your camera(s) does copy use less cpu than libx264 or libx265?

kevinGodell, 2016-12-30:
copy and libx264 both bounce around 10%, while libx265 is around 20%. I guess that would be expected since the stream is h264, and if we change codecs with video creation, it will be a higher workload for ffmpeg. I was just pointing out the crashing because i figured that was a slight structure problem with the ffmpeg command.

kevinGodell, 2016-12-30:
On a side note, as i was trying all of the various combinations of settings(attempting to make it crash), I realized that there are many settings which do not need to be set if you are using "watch only." You do toggle the visibility of some of them, but some are still visible such as file type, codecs, (and possibly width and height?). I hope you do not think I am trying to be critical, but I am definitely trying to find faults so that you can make a successful product.

moeiscool, 2016-12-30:
one of the main reasons why i made it open source and free is so that i can get that input :) It makes development faster when i don't have to notice everything for the final product :)

moeiscool, 2017-01-16:
it turns out this will always be a problem if i decide to include `copy` as an encoder. I really like `copy` because of how little CPU it uses. I am going to close this thread and add it to the wiki in a new page like "Known Issues" or something.

closed 2017-01-16 by moeiscool