moeiscool/Shinobi #14: JPEG Camera stream issue (Works fine in ZM)

knackrack615, 2016-12-30:
So i have a D-Link DCS-932LB and using the exact same settings that work on Zoneminder i get no stream on Shinobi

kevinGodell, 2016-12-30:
I had a similar issue with 2 of my cams. The problem for me is that zoneminder was still getting video from my cams and they were not able to make a 2nd connection. As soon as I turned them off in zoneminder, they worked in shinobi.

knackrack615, 2016-12-30:
Just tried disabling the camera in ZM.. no luck

Also my camera seems to have no issues streaming to multiple sources at the same time (I tried opening 3 tabs + ZM without issues)

moeiscool, 2016-12-30:
this is gonna sound dumb, but did you try disabling the timestamp in the options? @kevinGodell found some cameras didn't work with the timestamp option added.

knackrack615, 2016-12-31:
I can't find any option to disable/enable the timestamp

However what i did find was this in the error log https://scrgrb.xyz/i/index.php?picture=2016-12-31_02-07-56.png

moeiscool, 2016-12-31:
hmm socket error indicates a problem connecting to the camera
I just pushed a new commit.. it fixes the problem @kevinGodell was having with the timestamp crashing ffmpeg (I hope).

I have also added control functions.

kevinGodell, 2016-12-31:
was that commit from today? i see that the last commit posted is from yesterday.

moeiscool, 2016-12-31:
woops, didnt hit sync

moeiscool, 2016-12-31:
please check the newest commit and try "copy" as the vcodec with your camera.

moeiscool, 2017-01-01:
after much banter with @knackrack615 we found ffmpeg wasn't available for his version of Ubuntu, updated to 16.04 to correct to issue.

closed 2017-01-01 by moeiscool