moeiscool/Shinobi #17: OZone and zmNinja

pliablepixels, 2016-12-31:
Good stuff. 
A) if you consider adding  an API I can look at adapting zmNinja to work with it
B) It would be interesting to see shinobi be a front end for oZone (http://ozone.network) 

Just putting it out there :)

moeiscool, 2016-12-31:
I've already started on an API system, just need to secure it with keys. What options should do you need specifically?
I am guessing these for sure:
- get monitors
- get streams (you want mjpeg streams? Or can your client make use of socket.io?)
- get events by monitor
- edit monitors

pliablepixels, 2016-12-31:
we should probably have an offline chat. Skype sometime?

knackrack615, 2016-12-31:
I've made a server on Discord for Shinobi, we can talk there https://discord.gg/mdhmvuH

You can join without even downloading discord so its good @moeiscool :)

moeiscool, 2016-12-31:
Skype hurts my heart. I just got on discord, or if you want you can join the slack team.. but discord seems to be easiest

kevinGodell, 2016-12-31:
I would like to experiment with whatever streams you can give access to in an api. A few months ago, I was about to rebuild my desktop and mobile apps for zoneminder. I was playing with h263 and h264 encoded flv streams and reading them to my desktop app attempting to find a lower bandwidth solution to mjpeg. Bandwidth was very low on h264/flv, but the cpu on the server running zoneminder(2 quad cores) was taking a beating because of poor resource management, which was very discouraging.  

With my app connecting to zoneminder, I was able to decode jpegs, mjpeg, h263/flv, or h264/flv. I would like to access those streams from shinobi if possible. Also, sockets will be needed by an app or browser if you want to access more than the 6 stream limit set by some browsers and frameworks. I think zm ninja ran into that limit and had to resort to single jpeg refreshing, but I could be wrong on that.

moeiscool, 2017-01-01:
@kevinGodell technically a server can consume the stream too. When child.js (in development still) is used it does the ffmpeg process and pipes stream frames back to the main server then the handles it the same way it would if it were running the ffmpeg process on the same machine.

so if you can set your server up to consume through websocket then that would be pretty sweet... otherwise i can try and look into making it output to mjpeg.. but i may push that along and make room for more pressing features like motion, region motion, vehicle/license plate detection.

kzap, 2017-01-01:
is there a slack or discord only?

moeiscool, 2017-01-01:
there is a slack group, if you want you can message me on discord with your email and ill send you an invite.

I admit though I really like discord.

moeiscool, 2017-01-01:
@pliablepixels if you wanna hit me up on slack/discord we can talk API :) for now i think i'll close this thread.

closed 2017-01-01 by moeiscool

moeiscool, 2017-01-02:
Just set up an API, info can be found here:
https://github.com/moeiscool/Shinobi/wiki/API-Access

pennybuster, 2017-03-30:
Hi there! Sorry to hijack this thread. I'm a newbie but I want to contribute with a suggestion.
I think that Shinobi will benefit more from having a React Native app with this video component that supports exoplayer for Android (https://github.com/react-native-community/react-native-video/tree/master/android-exoplayer) than going the zmNinja route. 
Thanks a lot @moeiscool 
Greetings!