moeiscool/Shinobi #20: mysql Monitors small bug

kevinGodell, 2017-01-01:
It seems that if you have already created a camera and saved it, but then later change the Monitor ID, it will cause a new camera to be added to the database instead of updating the existing monitor. Also, when creating a new monitor, if using a period "." in the Monitor ID, will cause it to break and not show the live stream. I was just trying to use a camera ip address as the mid.

It seems like the mid should be an auto increment id, instead of something that a user can edit. I am no mysql master, but I seem to remember auto increment id as something I used to use often.

moeiscool, 2017-01-01:
next commit will have an inhibitor on the field. The reason why I did this is one not so noble. Originally when I was developing with zm my intention was to make a platform like mangocam. With a large number of users increments can go up fast and i suppose my ocd just finds it horrid .. the idea of it possibly reaching a maximum number and no longer able to create more rows.

so the camera is identified by its custom id + group key instead.

I am open to having my logic reassessed, just need a convincing reason why :P

kevinGodell, 2017-01-01:
On an earlier commit after your release announcement, I was unable to add monitors. So i installed phpmyadmin and went in for a look to see if i could add a monitor directly from that interface. phpmyadmin was complaining about the Monitors and Videos tables: "Current selection does not contain a unique column. Grid edit, checkbox, Edit, Copy and Delete features are not available " and it would not let me manually edit the entry unless i altered mid and tagged it "Unique." 

So, I am not saying that it will not work the way you have it(and frankly i did not want to bring it up and sidetrack you from your progress), but just trying to keep you informed so that you can keep building while some of us keep looking for bugs and try to offer suggestions. When I get some time to get setup and study node some more, I will offer some pull requests instead of just opening issues.

closed 2017-01-05 by moeiscool

kevinGodell, 2017-01-05:
Late reply. Didn't know you responded. I have the latest version of MySQL installed. It was a fresh installation of Ubuntu server 16  just for the purposes of testing shinobi. Events do get saved and I am able to add monitors via shinobi interface using the later commits. I don't know if uniquesness is necessary for database entries, I just thought it was the standard way.

moeiscool, 2017-01-05:
@kevinGodell it would be cool if could join us on Discord https://discord.gg/mdhmvuH

Well I guess I've never seen it not used, personally I don't think its needed when the time, group key, and monitor ID in combination  do this. When i was trying to make another program with ZoneMinder.. the event count got to 9000+ pretty quickly. The original purpose of working with it was to make a platform like mangocam.. and if i hope to have lots of users then it would soon become an issue if SQL doesnt reset on its own when it hits max number.