moeiscool/Shinobi #21: please add Hikvision DS-2CD2135F-IWS to supported camear to wiki

mrproper, 2017-01-02:
Manufacturer: HikVision
Model: DS-2CD2135F-IWS

Resolution: 2048x1536
Type: H264 (camera supports h265 and h265+ but they dont seem to work correctly with shinobi...green artifacts)
Stream URL: /Streaming/Channels/1?tcp
Save To: MP4 (I couldnt get webm to work)

moeiscool, 2017-01-02:
thank you for the entry :)

do any of your cameras save to webm? this was happening on another system. it turned out ffmpeg didn't install with libvpx (VP8/VP9 encoder).

perhaps trying to install the latest `libav-tools` will solve the issue?

If you want you can join me on Discord https://discord.gg/mdhmvuH and we can try and hash it out 1-1

moeiscool, 2017-01-02:
Camera added.

closed 2017-01-02 by moeiscool