moeiscool/Shinobi #22: using ffmpeg -f segment filter for record splitting

mrproper, 2017-01-02:
I've noticed that you're no longer/not using ffmpegs segment filter for splitting files over the cutoff period, is this intentional, looks like you're respawning ffmpeg after the cutoff...please correct me if im wrong. This would create a small gap between recording intervals.

moeiscool, 2017-01-02:
yah for some reason it kept crashing when i tried to use it.. so i setup a cutoff function by kill and respawn.

I would imagine it does.. depending on how fast your rig is it may just look like a flicker on the stream. On my macbook its a flicker with 4 cameras going to webm. (one of each type)

moeiscool, 2017-01-21:
next commit to master will include segmenting.

closed 2017-01-21 by moeiscool