moeiscool/Shinobi #27: ffmpeg doesnt like spaces

kzap, 2017-01-03:
maybe we can put double quotes around the output location

```2Df5hBE : bunny
FFMPEG STDERR
[NULL @ 0x7fa43700de00] Unable to find a suitable output format for '/Users/andremarcelo-tanner/Documents/Web' /Users/andremarcelo-tanner/Documents/Web: Invalid argument```

the directory im running from on OSX has a space in its name

moeiscool, 2017-01-03:
Will do in the next commit.

moeiscool, 2017-01-16:
ok so its been a while with this open and the reason i didn't add it is because it break h264 for some reason.. im not sure why. does it breakit for you?

moeiscool, 2017-01-28:
ok so the double quotes kept causing my ffmpeg instances to crash. I have instead added an option for this. You can choose to double quote the directory on a specified monitor. The option is in the recording section.

closed 2017-01-28 by moeiscool