moeiscool/Shinobi #28: application starts then aborts, fresh install Ubuntu 16.04

the4ndy, 2017-01-06:
I have a fresh install of 16.04 that I am trying to install Shinobi on. I used the Install.sh script and everything appeared to have worked fine. I run ``` node camera.js ``` and that says its listening on 8080 but then shortly after that (like before i can get to a browser and test it out) I get the following error
```
nodejs: src/unix/linux-core.c:747: read_times: Assertion `num == numcpus' failed.
Aborted
```
Any assistance is greatly appreciated. Thanks

moeiscool, 2017-01-06:
It seems like a problem with either `os-utils` or `os` (plugin for reading hardware status).

In this version of camera.js i have disabled those functions. Please overwrite your your old camera.js with this one.

https://mega.nz/#!65o3TJAS!nnp6y6GN3rJh5kNTRNrRGP3m6J3IMvQVPkC18oTbDRo

moeiscool, 2017-01-09:
Have you gotten up and running? I sure hope so. If not join me on Discord I'll be able to give you quicker help :) https://discord.gg/mdhmvuH for now I'll close this thread.

closed 2017-01-09 by moeiscool

paladine-raistlin, 2017-01-24:
I have same issue with latest on Ubuntu 16.04 LXC, Proxmox. I tried your modified file and that fixes issue. 
Thank You

moeiscool, 2017-01-24:
please note that the modified file is very old. please join me on slack so we can work out a better solution.

moeiscool, 2017-01-24:
@paladine-raistlin and I have found the solution. Updating your node.js to 7.2.0(+) eliminates the issue.

Updating Node.js can be done in the following way
```
sudo npm cache clean -f
sudo npm install -g n
sudo n stable
```