moeiscool/Shinobi #29: use shinobi as an active relay

turbomettwurst, 2017-01-09:
I am currently trying to get video surveillance running at a location that is behind a nat router which is beyond my control. 
I therefore cannot actively pull the video stream from the camera.
Could shinobi be adapted to run twice, once on premise, working as a proxy that pulls the stream from the camera and actively transmits it (via rtsp?) to a second instance running somewhere publicy accessible?

moeiscool, 2017-01-09:
Shinobi has this feature, it's currently not updated to work with the new scheme, child/child.js does this.

Child.js runs on another machine (as many as you want). It does the ffmpeg command, streams any video to master server then the master sends it out to anyone watching, creates videos(if monitor is recording), and sends the videos it creates back to the main server for storage (removes video file after transfer).

The original purpose was not to use this on multiple networks though. It was meant for multiple nodes to be in the same network working together. a load balancer I guess. I have never tried to use it over the internet. If these two servers can maintain their socket connection between each other then it should be okay. Otherwise I'm not sure what kind of issues you might have if for a moment the child.js is not found during a process.

turbomettwurst, 2017-01-09:
Ah that sounds promising, thx a lot!
as for the "real world stability", that is something only real world trial and error can answer ;-)

moeiscool, 2017-01-10:
hmm i updated the file but its not working the way its supposed to.. I'm going to leave this open until i get it fixed.

turbomettwurst, 2017-02-26:
Sorry for the late feedback.
signed up at gitlab, couldn't find any code.
I did however pull your master from here and tried to understand it..
what host belongs into conf.child.json? the remote, public shinobi instance? How do i get the remote instance to listen?

moeiscool, 2017-02-26:
child.js is using socket.io-client to communicate with camera.js. host should be localhost if child.js is on the same server and port matches the websocket listening port in conf.json for camera.js

moeiscool, 2017-04-11:
since this thread opened some of the posting rules changed. now suggestions go in the slack #suggestions channel or asked about on reddit.

closed 2017-04-11 by moeiscool