moeiscool/Shinobi #30: monitor video not fit for purpose

ITwrx, 2017-01-12:
hi,
first of all, thanks for shinobi. It looks very promising.
I installed shinobi on arch and it is working fine but the video from the cameras is too poor to be useable. There are too many artifacts and too much streaking. I read the faq where you said you scaled the video down to 640x480 but my video playback is even worse than that, it seems. I can run "ffplay 'rtsp://user:password@192.168.1.5:554' -analyzeduration 0.1" and the video is at native cam resolution or manually scale down(adjust the size of) the windows (make my own psuedo monitoring grid) and the video stays very nice. no artifacts or streaking either. I'm not sure what the issue or solution is, but the monitor video is too poor to be used for it's intended purpose. I would also say that getting monitor video working as well as ffplay might ought to be fleshed out prior to starting on motion detection, as users could set motion in the cam's gui while monitoring in a unified interface (shinobi) while motion in shiinobi was worked on. IOW, motion is not an absolute deal breaker, like monitoring is(at least for me).
thanks

moeiscool, 2017-01-12:
Yah sometimes when my cameras reconnect ill get streaking or green artifacts. The pixelation is due to the resizing. Right now I am actually working on having the image resizing done outside ffmpeg and in the node.js app before it gets sent. I hope for this to deal with many problems we are seeing in the live stream.

what version of FFMPEG do you have? version of arch? If you by chance get it going on Ubuntu 16.04, i'd like to know if your cameras behave the same as they do in Arch.

I'll leave this open until i get the resizing adjustment made... hopefully it will deal with the artifacts and such.

ITwrx, 2017-01-12:
good news. i think that approach will probably work, too. i'm running ffmpeg 1:3.2.2-1. Arch doesn't have versions, as it's a rolling release, but i'm running linux 4.8.13-1 and if you want any other package versions, just let me know. Also, you might consider testing against arch since it's closer to upstream in release timeframe and "purity", instead of only using slower/downstream distros with sometimes modified packages. I'll let you know if i get an ubuntu box set up.

Thanks

ps. where do you want feature requests? here or reddit?

moeiscool, 2017-01-13:
heh your ffmpeg is newer than mine. I'm on 2.8. Did you have to compile ffmpeg yourself? Hit me up on Discord https://discord.gg/mdhmvuH . Sweet I'll give Arch a go in VirtualBox. My macbook is happy on Ubuntu Server :D 

I prefer to have them made in the #suggestions channel in Discord, but you can post them on Reddit if you want.

ITwrx, 2017-01-14:
no, that's just the current stable version of ffmpeg. Arch packages are binary by default. You can build from source if you want to, but you don't have to for official packages. BTW, you could also use LXC for the virtual arch. LXC is tremendous. :) (hardly any overhead, etc.) 

Thanks for the invite for Discord but i'll have to think about whether i'm going to sign up there or not (another closed source saas app that owns everything i do while using their service/datamining operation...). Will have to look at reddit's TOS and PP as well.

thanks

moeiscool, 2017-01-14:
I just found a way to make slack do invites automatically. https://shinobicctv.herokuapp.com/ if you wanna do slack join me there :) I am going to be updating all the Community Chat references to match slack. https://shinobicctv.slack.com

moeiscool, 2017-01-16:
not sure if we talked on slack yet.. but is this streaking happening on all types of cameras?

ITwrx, 2017-01-16:
no, we haven't. i will use branded nick if i sign up. I haven't tested with other cams, but these cams work perfectly with ffplay command, which is now "ffplay 'rtsp://user:password@192.168.1.5:554' -analyzeduration 0.1 -x 960 -y 520 -window_title front_cam". that's what i'm using as my montiors right now. They go 24 hours a day @1080p over the lan without **any** pixelation. I'll try to add my usb web cam and old foscam to shinobi if i get some time.

moeiscool, 2017-01-16:
hmm i just tried adding that in my pipe and it didn't work. do you have an example ffmpeg string with your settings?

ITwrx, 2017-01-16:
yeah, ffplay is the player for ffmpeg. i think it needs a gui so it can open a window. this "ffmpeg -i rtsp://user:password@192.168.1.5:554 -acodec copy -vcodec copy /home/user/cam_stream.mp4" just worked well for me, but of course it's recording to a file. my js is too weak or i'd know how to get it into a browser. I suspect that the reason the above ffmpeg command worked well is because it's not trying to transcode anything. It's just copying to a file. You may have to/want to separate the monitoring logic from the recording logic in shinobi? no transcoding for monitoring and transcoding for recording. 
also, have you seen this?
https://github.com/SpecForge/html5_rtsp_player
one way to get the rtsp videeo into the browser for montoring without overhead and lag.

moeiscool, 2017-01-17:
when you record do you get streaks or only on your streams?

ITwrx, 2017-01-17:
i just tested the recording. i used the "copy" settings for vid and audio so it should be similar to the ffmpeg command i posted previously. the video seems to be at the proper resolution(it's in night mode so it's not super easy to tell) and was largely usable. It had **momentary**, horizontal streaks 10 times during the 6 minute video and one horizontal "curtain" of streaks that only lasted 2-4 seconds. of course, that would be bad if it were capturing something important while that happened. there are ffmpeg errors but these are also present when playing with ffplay so i don't know how pertinent they are. I can report them as seperate issues if you want. Also, the video shinobi just recorded is shown twice in the right hand pane of the gui(same vid file size shown). I can also report that, if it's not already a known issue or something. I would say that the stream is streaking a little more often than the recording. the most noticeable difference is the extreme pixelation/low res of the stream vs the recording.

moeiscool, 2017-01-17:
when you do mjpeg do you get streaking? im not sure if this ffmpeg problem or just how im using it... because even the image stream are images made by ffmpeg. Its piped off the ffmpeg process thats recording/watching

yes i noticed that sometimes the items on the side would duplicate. I am going to rework how that is written on the side.

moeiscool, 2017-01-17:
can you check out the newest dev commit? i think the streaking is gone.

ITwrx, 2017-01-18:
is the page element being scaled via css or the video source via ffmpeg?

moeiscool, 2017-01-18:
both. **Stream Scale** can be changed in the newest commit.

ITwrx, 2017-01-18:
i ran "chmod +x UPDATE.sh&&./UPDATE.sh" and it didn't have pm2. installed pm2 and reran. It runs but pm2 doesn't find instance of camera.js. try to run "node camera.js" and it gives "Error: Cannot find module 'ejs'. ". so, i wasn't able to test. If you don't know what that error is about i can try and start from fresh install later.

moeiscool, 2017-01-18:
oh sorry, please run
`npm install ejs`
I added ejs in the package.json but it seems its not getting installed on some machines.

moeiscool, 2017-01-18:
ahh i see what happened. I forgot to make the package.json get updated as well when the update function is run. It will not give such errors in the future.

ITwrx, 2017-01-18:
thanks. i got it updated. i see your new "stream scale" setting but i still don't understand your previous answer regarding this. When i play my cam stream with ffplay the resolution of the video is 1920x1080 even though i tell it to output to 960x520 window. this results in a very clear stream. I thought you were going to only scale the page element via js/css and leave the video source at native resolution? maybe you did? It doesn't appear that Shinobi is outputing 1920x1080/native video source in scaled page element. It starts flashing when i use 1280x720 and flashes with no video at 1920x1080 "stream scale" setting. This leads me to believe there is some problem as ffplay doesn't struggle at all at 1080. The stream vid in shinobi is still more pixelated than ffplay/ffmpeg record command with occasional, brief streaking. also, the color is a little off. there seems to be a purple hue to it. 

I see you fixed the double file on the right side issue. please note that the double file is still shown in stream element overlay.

moeiscool, 2017-01-18:
Yes unfortunately the this is because of the way I'm streaming the image to the browser. Right now it base64 encodes each frame then sends it over websocket. When the image is too large the client doesn't have enough time to render it that's why it flashes.

 I wanted to use ffserver for webm/MP4 streams but it is deprecated and being removed entirely in the next major release, if not already removed.

I'm open to any suggestions. I can add other streaming methods for sure, the code is very easy to work with... so far 😅

ITwrx, 2017-01-18:
yeah, i saw the ffserver situation. i intended to make my own web gui for my cams until i learned that browsers don't support rtsp and ffserver is being shuttered. is this project (https://github.com/SpecForge/html5_rtsp_player) doing the same thing as you or slightly different? 
besides that the only way i know of is to use webrtc which sounded like a bunch of "fun" for me but that's b/c i currently shy away from js for stuff more complicated than modifying page elements in html, etc. It might would be more feasible/easy for you.

ITwrx, 2017-01-18:
i see they (html5_rtsp_player) are using some mysterious websocket proxy software but i wonder if one could just use nginx for that part of the flow.

moeiscool, 2017-01-18:
wow that really is some sort of sorcery... i'm going to wander this jungle for a little while and see if i find any treasure.

ITwrx, 2017-01-18:
:) also, i don't know what usefulness might could be strangled out of this... 
https://www.npmjs.com/package/ffplay

ITwrx, 2017-01-18:
and this...https://github.com/yomguy/stream-m

ITwrx, 2017-01-18:
on more...https://bgrins.github.io/videoconverter.js/

ITwrx, 2017-01-18:
last crazy method? http://phoboslab.org/log/2013/09/html5-live-video-streaming-via-websockets

moeiscool, 2017-01-18:
if you find more like stream-m and **last crazy method** please post them.

im not sure how to use the ffplay node library, it looks like its only for simple stuff and the videoconverter.js i guess can be used for something unrelated. Its actually ffmpeg for the browser.

ITwrx, 2017-01-18:
this is one example of a webrtc method. 
http://www.kurento.org/blog/interoperating-webrtc-and-ip-cameras
https://github.com/lulop-k/kurento-rtsp2webrtc

ITwrx, 2017-01-18:
you could possibly transmux rtsp to hls:
http://stackoverflow.com/questions/35013087/stabilize-ffmpeg-rtsp-to-hls-transmuxing
https://github.com/dailymotion/hls.js
(or maybe mpeg dash)

ITwrx, 2017-01-18:
http://blog.wmspanel.com/2015/05/rtsp-transmux-hls-rtmp-dash-mpeg2ts.html

ITwrx, 2017-01-18:
you can theortically stream rtsp with gstreamer and pull into the browsr via video tag, but i haven't got a working gstreamer command on my machine yet.
https://gist.github.com/shocrunch/0c038321d05586841897

moeiscool, 2017-01-19:
you'll be glad to hear the flickering has been dealt with. 1920x1080 and up still have some issues but I am sure i'll get them figured out soon :D I adapted a method off this http://phoboslab.org/log/2013/09/html5-live-video-streaming-via-websockets

instead of `img` tags, i used `canvas`.

ITwrx, 2017-01-19:
great! glad i could help in some way. I just updated and i'm not getting any flickering but the stream is still unfit for purpose. It's still too pixelated and blurry. It still looks like the source is scaled in ffmpeg to 640x360 or something. Not just the page element, but the video source. I don't see any way it can be scaled with ffmpeg to a low res and not look like one is underwater.

moeiscool, 2017-01-19:
newest commit battles pixelation with great finesse. Thank you again! the canvas was the answer for many issues. I have yet to see the streaking.. but it probably wont be fixed by this.

after reading jsmpeg like a good book it turns out that library is just for reading mpeg arraybuffer data. sadly that method excludes audio... but what it also means the method i learned to grasp is in essence the same thing, just less processing for the server and client since it uses my old method but just draws it to canvas rather than an image dom.

ITwrx, 2017-01-19:
just updated and tested. Big improvement! I'm able to set them all to 1280x720 and it mostly looks to be 720p. There is some streaking and some pixelation but it is definitely getting there. ffplay is still better(and in 1080p), just as a comparison. Please keep me posted on any further improvements.

moeiscool, 2017-01-19:
can i get another comparison please? your cameras have nice detail even at night time :D

ITwrx, 2017-01-19:
ffplay: https://i.imgsafe.org/053a2cb241.png
shinobi: https://i.imgsafe.org/0538cddad9.png
i'll upload some during the day tomorrow too.

moeiscool, 2017-01-19:
newest commit has a **Stream Quality** option

moeiscool, 2017-01-19:
Hmmm it seems if i make the quality and scale too big the image loads only partly. Like a dial up connection. It seems to behave in the same manner as jsmpeg though... maybe they expected people to only use it with webcams.

ITwrx, 2017-01-19:
It looks better than it used to, and could actually be used now, but users won't get the res they expect from their cameras. I was not able to set min above the default HxW of 640x480, nor go below 10 in stream quality.
shinobi: https://i.imgsafe.org/10a7075ddc.png
ffplay: https://i.imgsafe.org/10a982bd08.png

ITwrx, 2017-01-19:
it seems that ffplay is 5-6 sconds behind shinobi! Maybe shinobi could have the option to trade latency for stream size+quality? If i could bump up HxW and stream quality at the expense of latency i would be happy with that. If someone needed real time they could prioritize that. Of course, real time and native res would be ideal, but until that's possible, this would be perfectly acceptable in my opinion.

moeiscool, 2017-01-19:
that's interesting. Is there anything that shows you the ffmpeg command ffplay is using when it does the video? I am curious know why it has a delay.. like what options is it using?

Is there a debug console? that shows this information or something?

ITwrx, 2017-01-19:
well, after restarting ffplay th dlay is only 1-3 seconds more than shinobi. previous incantation had been running for a couple of days so maybe it fell further and further behind. still, there is a couple of seconds to play with. terminal output follows.

ffplay 'rtsp://user:password@192.168.1.8:554' -analyzeduration 0.1 -x 960 -y 520 -window_title back_yard
ffplay version 3.2.2 Copyright (c) 2003-2016 the FFmpeg developers
  built with gcc 6.2.1 (GCC) 20160830
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-avisynth --enable-avresample --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libass --enable-libbluray --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libschroedinger --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-netcdf --enable-shared --enable-version3 --enable-x11grab
  libavutil      55. 34.100 / 55. 34.100
  libavcodec     57. 64.101 / 57. 64.101
  libavformat    57. 56.100 / 57. 56.100
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  3.100
  libpostproc    54.  1.100 / 54.  1.100
Input #0, rtsp, from 'rtsp://user:password@192.168.1.8:554':
  Metadata:
    title           : Session streamed by "preview"
    comment         : 
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(progressive), 1920x1072, 10 fps, 10 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: aac (LC), 16000 Hz, mono, fltp

moeiscool, 2017-01-23:
please try the newest commit of the dev branch https://github.com/moeiscool/Shinobi/tree/dev i think it addresses streaking. since this is the dev branch please back up your currrent `web`folder and `camera.js` file... or if you know which commit they are from just download them if need be.

What i had was the pixel format hardcoded. I now let ffmpeg choose what it should be based on other settings.

moeiscool, 2017-01-23:
nevermind streaking still occurs, just less. dangit

moeiscool, 2017-01-24:
please try `-rtsp_transport tcp` in the *Custom Input Flags* box.

ITwrx, 2017-01-24:
will do, hopefully tonight. use the dev tree still? in the past i was just running the update script you provided. very convenient, btw. my cams only have max bit rate setting instead of min or exact, but there's every indication that they use whatever you give them, as i had to turn the setting down to 3k or so.

moeiscool, 2017-01-24:
using your current install should be okay if it has the *Custom Input Flags* box.

ITwrx, 2017-01-25:
just ran update and tested. no streaking. still a little more pixelated than ffplay. i currently have the stream set to 720p in shinobi.

moeiscool, 2017-01-25:
`-rtsp_transport tcp -analyzeduration 10000000 -probesize 10000000 -use_wallclock_as_timestamps 1`

these are the input flags that @kevinGodell uses with his 1280x720 H.264 Camera.

maybe this can handle of that pixelation?

ITwrx, 2017-01-25:
ok now i'm getting what looks to be true 720p, though i'm still hoping for 1080p as my cams will do up to 1440p. i only need 1080p though. i used "-rtsp_transport tcp -analyzeduration 0.1", framerate of 8 and quality of 12. i had accidently set quality to 23(probably due to updates?) so that was the pixelation issue. If i try to set them all to 1080p it crashes firefox. i'll see what i can get away with in chromium.

moeiscool, 2017-01-25:
hm, i didnt set anything to change it and the updates dont run any sql updates yet :/ was there anything else with a incorrect value? any of your other cameras with incorrect values? you're not the first person to say a field was different than what they remembered. which i find really strange. Time to bust out the scooby snacks.

when you set one of them to 1080p, do you get the whole picture?

ITwrx, 2017-01-25:
chromium acted similarly. the video element is never drawn all the way to the bottom when set to 1080p. also, on both browsers it still looks as if the cams are looking through an extra dark (slightly purple) tinted plastic lense. IOW, ffplay is lighter and more color accurate.

ITwrx, 2017-01-25:
i'm not sure if there were any other fields with default values that i thought i had set but it definitely was more than one camera and i want to say it was all of them.

ITwrx, 2017-01-25:
btw, idk if it's useful to you but evidently firefox OS supports rtsp.

moeiscool, 2017-01-25:
yah i hope to get rtsp output shinobi soon. :) since the streaking is down im gonna close this thread. you can also try `http` instead of `tcp`. i'll add a dropdown for this option in a future commit.

ITwrx, 2017-01-25:
i just had to turn my settings down to 960x520 with quality of 18 with fps=8 to get shinobi to render the monitors' video during the day time. Anything above that and it won't draw the full video and/or crashes firefox. during the night it will do 720p with 14 quality. do you want me to open another issue or do you want to reopen this one?

kevinGodell, 2017-01-25:
I am curious. When using ffplay, how do you view your streams? Do you use an app or do you view them through a browser? If ffplay is the comparison, then we must know why it is currently more capable than shinobi. Thanks

ITwrx, 2017-01-25:
hi,
assuming you have a DE(desktop environment. i'm using gnome 3), if you enter the ffplay command in my OP (substituting your user and pass) into a terminal it will automatically open up a window with the stream in it.

[22 more replies]

closed 2017-02-03 by moeiscool