moeiscool/Shinobi #32: add Wansview W3 (X-Series) to compatibility list

jm93944, 2017-01-16:
My Wansview model W3 is connected and working. I have a couple other models I will test soon also. 
Using these settings:
H.264
 MP4
libx264
RTSP
Path,   ip/live/ch0
port,     554

I have Shinobi running on my Lenovo Yoga 11E laptop, with Linux Mint 18.1. Followed installation instructions for Ubuntu 16.04.
So far this was much easier than setting up zoneminder, looking forward to seeing this project mature.
Thank You!

moeiscool, 2017-01-16:
it has been added, i also added the laptop to supported systems. Very much appreciated! :D

closed 2017-01-16 by moeiscool

jm93944, 2017-01-21:
I have also been testing Shinobi with my Wansview cameras, models W2, and Q1.

I could not get the PTZ function to work on the Q1 model camera.
When I enter this URL in the browser it works.
"ip address"/hy-cgi/ptz.cgi?cmd=ptzctrl&act=right|left|home|up|down

Thanks again!

moeiscool, 2017-01-21:
Hey @jm93944 please join me on slack to diagnose your camera's PTZ :)
https://shinobicctv.herokuapp.com/