moeiscool/Shinobi #35: Number of Days to keep Videos - Not working

vLX42, 2017-01-17:
The videos are not getting deletede after x days.

I have set it to 3 days, but I now have recording for 6 days

moeiscool, 2017-01-17:
Do you have `cron.js` running? I really should write something about it lol, sorry about that.

vLX42, 2017-01-18:
I tried to run cron.js, it deleted all events in the database but non of the files. Maybe something with dateformat or OS. I'm running Ubuntu 16.04.1 LTS and MariaDB (Mysql fork)

moeiscool, 2017-01-18:
just pushed an update for the cron file.

jm93944, 2017-01-21:
I had the same problem the videos were removed from Shinobi, but files were not deleted. I just updated and will see if that fixes it.

moeiscool, 2017-01-21:
I think I'll keep this open until I know cron actually works for everyone, I just closed it because I got no response :(

vLX42, 2017-01-24:
Still having the same problem. If i can do anything to help please say so.

moeiscool, 2017-01-24:
Can you join me on slack? We can easily talk about what you're experiencing. I may just rewrite the cron file...

moeiscool, 2017-01-28:
i have reverted the cron to an older version. currently It will ignore orphaned files. I still need to create a method for this.

closed 2017-01-28 by moeiscool