moeiscool/Shinobi #44: PTZ setttings not working with Wansview Q1

jm93944, 2017-01-21:
I could not get the PTZ function to work on the Wansview Q1 model camera.
When I enter the URL directly in the browser it responds.
http://[camera ip address here]/hy-cgi/ptz.cgi?cmd=ptzctrl&act=right (left, up, down, home also work)

No response when trying the same in Shinobi. Any suggestions?

moeiscool, 2017-01-21:
When you open the control URL in your browser directly does your camera behave the way you want? (Move left or right, etc.) are you able to make it move without Shinobi and with only the control URL from a browser?

The way the PTZ works is that it sends a get request to the specified path.

moeiscool, 2017-01-21:
Just to let you know there are many learned people on the slack community chat :) We may not be in time zones where we have a convenient chances of speaking to each other but some members are awake during the hours I sleep and would gladly give some input if they are on.

https://shinobicctv.herokuapp.com/ is the invite page

jm93944, 2017-01-21:
yes, 
If I paste the URL into the browser the camera returns to home/center. 
http://192.168.1.144/hy-cgi/ptz.cgi?cmd=ptzctrl&act=home

So most other cameras need a get request, but maybe mine is not typical?
I'll ask on slack, thanks!

moeiscool, 2017-01-28:
hey was wondering if you got it working?

jm93944, 2017-01-30:
I did not. Let me know if you have any ideas.

moeiscool, 2017-02-03:
this might sound dumb but can you check that your version of node.js is at least 7.2.0 or higher?

`node -v` or `nodejs -v`

jm93944, 2017-02-09:
Looks like Linux Mint 18 is only giving me 4.2.6 
I updated it, but lost my remote connection to the machine. I'll let you know if the update helped once I get my connection back, might be a few days.

jm93944, 2017-02-12:
Couldn't get it to update past 6.9.5 from repos in linux Mint 18.1
 I'm compiling 7.5.0 now...

jm93944, 2017-02-12:
now v7.5.0 but still just get the same message.

2Df5hBE : Q1cam
Control Error

where in the shinobi js code can I find those control functions?

moeiscool, 2017-02-12:
Line 959 in camera.js is the handler for control functions.

tfohlmeister, 2017-03-21:
Same problem here. I am also wondering: Do I need to provide a full URL or a relative URL for each control command? E.g. `http://[camera ip address here]/cmd_up` or just `/cmd_up`?
In the example video for foscam you are setting a relative URL. But what server and port are used by Shinobi in this case? What if the control server port differs from the streaming port?

moeiscool, 2017-03-21:
There was a user who posted on the disqus thread with following message. I haven't tried the fix myself yet. I will be adding the appropriate fixes in the next few days. Just working on filters and cron at this time.

**SzaZo :**
*Hi,*

*I found if you are using RTSP, the control functions are not working because the camera.js wants to send through RTSP the control commands. Is it right? If i changed at case monitor section this `request({url:d.base+d.m.details['control_url_'+d.direction+'_stop'],method:'GET'}` to `request({url:d.m.details['control_url_'+d.direction+'_stop'],method:'GET'}` **(Without the d.base)**, and set at the control url the full control url with username and password, it works. is it this helpful?*

tfohlmeister, 2017-03-21:
Thanks @moeiscool, this does indeed work. For anyone stumbling upon this: Make sure to remove `d.base` from both the stop request (`'control_url_'+d.direction+'_stop'`) as well as the movement request (`'control_url_'+d.direction`).

moeiscool, 2017-04-04:
newest commit has an added feature allowing the user to put a custom base URL.

Considering the workaround works and the added options are to use this workaround : i'll deem this thread finally closed. cheers!

closed 2017-04-04 by moeiscool