moeiscool/Shinobi #52: Streams are corrupted and improperly scaled

ShanDestromp, 2017-01-22:
I'm setting up Shinobi in an Ubuntu 16.04 LXC on a base of Proxmox 4.4 (Debian 8).

My cameras are Reolink RLC-420's, using ONIFV (h.264 RTSP, port 554).  Stream quality is set to default however both my stream width and capture width are set to 2560x1440 as they're 1440p 16:9 cameras

The problem is that Shinobi seems to expect all cameras are 4:3 and not only squashes the view but seems to arbitrarily crop the image.  If I minimize the camera list and feed list (whatever you call the right hand side); the display box will adjust to 16:9 however the image itself is still cropped, and then stretched to fit.  See attached pictures for a comparison between my normal playback software and Shinobi.

[image]
[image]

Furthermore, Shinobi seems there is significant image artifacting (as in, chunks of blockyness / black that eventually clear up).  Reducing the stream dimensions (eg to 720p) *help* but don't solve this.

Below is the only error messages in shinobi's logs.

It is not a bandwith/HDD speed issue; this is all internal to a dual GigE LAN with a 24TB array so my read/write speeds are plenty; furthermore I can use ffmpeg to pull a stream down directly and it plays back fine

```
2Df5hBE : fEMOXc36mM
FFMPEG STDERR
[h264 @ 0x193d740] cbp too large (732) at 51 65 [h264 @ 0x193d740] error while decoding MB 51 65
2Df5hBE : fEMOXc36mM
FFMPEG STDERR
[h264 @ 0x1f5ee40] negative number of zero coeffs at 148 86 [h264 @ 0x1f5ee40] error while decoding MB 148 86
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x2462c60] out of range intra chroma pred mode at 127 67 [h264 @ 0x2462c60] error while decoding MB 127 67
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x226eaa0] negative number of zero coeffs at 153 53 [h264 @ 0x226eaa0] error while decoding MB 153 53
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x24aa420] negative number of zero coeffs at 65 52 [h264 @ 0x24aa420] error while decoding MB 65 52
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x226eaa0] dquant out of range (-344) at 113 53 [h264 @ 0x226eaa0] error while decoding MB 113 53
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x226eaa0] negative number of zero coeffs at 119 53 [h264 @ 0x226eaa0] error while decoding MB 119 53
2Df5hBE : 0iNo6JtpxZ
FFMPEG STDERR
[h264 @ 0x226eaa0] negative number of zero coeffs at 158 53 [h264 @ 0x226eaa0] error while decoding MB 158 53
2Df5hBE : fEMOXc36mM
FFMPEG STDERR
[h264 @ 0x1f5ee40] mb_type 50 in P slice too large at 125 72 [h264 @ 0x1f5ee40] error while decoding MB 125 72
2Df5hBE : fEMOXc36mM
FFMPEG STDERR
[h264 @ 0x1ecfec0] P sub_mb_type 8 out of range at 116 76 [h264 @ 0x1ecfec0] error while decoding MB 116 76

```

moeiscool, 2017-01-22:
when your monitor first starts it will output a ffmpeg command, can you please post that here.

ShanDestromp, 2017-01-22:
Shinobi was updated (with UPDATE.sh) a few minutes prior to the bug report (so after your last commit as of now), as the cameras didn't seem to work at all when I first added them.  I'm not sure where shinobi outputs any ffmpeg commands, I'm starting it with `pm2 start camera.js` per the docs and there's nothing in shinobi I can see nor in the /home/USER/.pm2/logs file; however I can get a stream manually (EG not through shinobi) via ffmpeg with

`ffmpeg -i rtsp://USER:PASS@IP_ADDRESS554/h264Preview_01_main -acodec copy -vcodec copy /path/to/output.file`

Camera Settings; basically everything is as default except as needed for my specific camera.

[image]

Including this one to show how the image is further corrupted (loses the whole bottom 1/3rd of the image.
[image]

moeiscool, 2017-01-22:
the orange button next to the blue snapshot button will open a logs pane on the camera. when you first start the monitor you will see "monitor starting" with it the ffmpeg command.

this is because the is not loading fast enough before the next frame comes in. You can adjust your stream quality or resolution to fix this. its sadly one of the caveats right now, if i can somehow compress the image more maybe it will load quicker... but for now the solution is scaling down or lowering the stream quality.

I hope to setup webm/mp4 streaming to the browser as an option instead of frames drawn by canvas.

closed 2017-01-25 by moeiscool

ITwrx, 2017-01-25:
this issue raised more than one issue. part of this issue was that the images are being cropped when minimizing or maxing side panes. as far as i know that is not fixed. should i open a seperate issue for the cropping problem or do you want to reopen this issue.

moeiscool, 2017-01-25:
We can discuss it in issue #30 , I close down this issue because I wasn't getting any replies from Shane, I assumed he wasn't having anymore issues.

ITwrx, 2017-01-25:
ok, i guess but the only part of this issue that correlates to #30 is the image artifacting @ShanDestromp  mentioned. the rest, and core of this issue, is a seperate matter. namely the 4:3 aspect ratio, stretching and cropping, which i was going to report until i saw this issue. #30 is all about quality.