moeiscool/Shinobi #57: Shinoby crashes after 1/2 days of use

Bonnee, 2017-01-27:
I discovered Shinobi after some ZoneMinder headache and hell if that's awesome! Good job devs!

But now I'm stuck with this issue that crashes Shinobi entirely (all the ffmpeg streams and the web ui).

Looking at the logs that's all I can found:

```
$ cat ~/.pm2/logs/camera-error-0.log
TypeError: undefined is not a function
    at Object.s.ramUsage (/home/bonnee/Shinobi/camera.js:1422:12)
    at /home/bonnee/Shinobi/camera.js:1427:45
    at /home/bonnee/Shinobi/camera.js:1418:10
    at ChildProcess.exithandler (child_process.js:645:7)
    at ChildProcess.emit (events.js:98:17)
    at maybeClose (child_process.js:755:16)
    at Process.ChildProcess._handle.onexit (child_process.js:822:5)
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
uncaughtException [TypeError: undefined is not a function]
...
```
and the log goes on like this for hundred of lines.

Are there more detailed logs or can you provide me a more verbose build which I could run? I'm willing to help solving this.

I'm running this under Debian 8

moeiscool, 2017-01-27:
It appears the code stopped somewhere before it could define this function (which is at the end of the file).

A few questions. Did Shinobi run correctly prior to this? Were there any cod modifications you made after installation? What interactions in the platform do you recall doing with Shinobi before finding it this way. Did the demo camera `bunny` work at all, could it record?

Please check that your install can connect to SQL.

If reinstalling from the master branch does not work please try the dev branch.
https://github.com/moeiscool/Shinobi/tree/dev

I will work on making it more verbose for situations like this one.

moeiscool, 2017-01-28:
just went through this with two other users, but i haven't experienced anything myself. Please try the latest commit.

Bonnee, 2017-01-30:
I updated to the latest version and, after a day it crashed again. The log still looks the same as before [camera-error-0.txt](https://github.com/moeiscool/Shinobi/files/740192/camera-error-0.txt).

To answer your questions:
* I noticed the issue a few days after I installed Shinobi, so I can't tell if it ever ran correctly;
* I made no modifications to the code, I only set the videos path in conf.json and applied the right permissions to that folder;
* I only used to stream a camera or two for 5 minutes or so and, after some hours, I tried to reopen the streams but I got a connection error (the entire UI wouldn't load and the processes on the machine were all dead);
* Bunny did non work probably beacuse I had a missing dependency that I found only after deleting bunny.

I can succesfully connect to MySQL.

I'll try to use the dev branch now.

Bonnee, 2017-01-31:
I noticed now it crashed again about 2 hours ago, but the UI was fine.

The log still contains the same `undefined is not a funcion` spam.
At the top I see this:
```
TypeError: undefined is not a function
    at Object.s.ramUsage (/home/bonnee/Shinobi/camera.js:1422:12)
    at /home/bonnee/Shinobi/camera.js:1427:45
    at /home/bonnee/Shinobi/camera.js:1418:10
    at ChildProcess.exithandler (child_process.js:645:7)
    at ChildProcess.emit (events.js:98:17)
    at maybeClose (child_process.js:755:16)
    at Socket.<anonymous> (child_process.js:968:11)
    at Socket.emit (events.js:95:17)
    at Pipe.close (net.js:465:12)
```

I noticed that this exception is thrown right when I start Shinobi and periodically all the other exceptions get thrown. So I guess It's a slow death

moeiscool, 2017-01-31:
Frankly i think there is something wrong with your setup. Can you try the latest commit? If that does the same thing please try updating your Node.js.

What version is it currently? you can find this info by typing `nodejs -v` in terminal. You must have node.js 7.2.0 or higher

Bonnee, 2017-01-31:
Ok, maybe this is the problem: I have Node 0.10.29 installed, which is Debian's default.
I'll try to update whenever I can. Will report back.

Bonnee, 2017-01-31:
Updated node and logs are clean now. I feel confident about this.
Thank you for the attention!

Keep up the amazing work!

closed 2017-01-31 by Bonnee

moeiscool, 2017-01-31:
I'm sorry that took 4 days for me to suggest. Just know that I will be able to help others with more ease :) Thank you for trying Shinobi!