moeiscool/Shinobi #71: Cron deletes db entries every ~2 hours

Bonnee, 2017-02-15:
During the last week I noticed that all recordings older than 2 hours disappear from the UI. I also noticed that all recordings get cleared when I restart Shinobi.

camera-out-0.log shows this whenever the recordings disappear:
```
CRON :  { f: 'did',
  msg: '1 SQL rows with no file deleted',
  ke: '2Df5hBE',
  time: '2017-02-15T13:24:03.989Z' }
```
The log states that rows with no file are deleted but all the files are actually in their location and they're not corrupted or anything.

I have set "Number of Days to keep Videos" to 10 and Max Storage Amount in Megabytes to 950000 (I'm running a 1TB hdd that only stores recordings and db entries).

Bonnee, 2017-02-15:
Update:
After updating to latest commit, the recordings are still kept after 2+ hrs, but they are still cleared by cron after  a restart.

moeiscool, 2017-02-15:
Can you try the con file from the latest dev commit? Just the cron.js not camera.js or web folder.

Bonnee, 2017-02-15:
I forgot to say that after #57, I run on dev branch. Node version is 7.5.0

moeiscool, 2017-02-15:
gets oldest 10 events in your sql
```
SELECT * FROM Videos WHERE ke=`2Df5hBE` ORDER BY `time` ASC LIMIT 10
````

gets videos older than 3 days
```
SELECT * FROM Videos WHERE ke = `2Df5hBE`  AND end < DATE_SUB(NOW(), INTERVAL 3 DAY)
```

when you run these commands do they do what they are supposed to?

Bonnee, 2017-02-15:
Here's the output:
```
mysql> SELECT * FROM Videos WHERE ke=`2Df5hBE` ORDER BY `time` ASC LIMIT 10
    -> ;
ERROR 1054 (42S22): Unknown column '2Df5hBE' in 'where clause'
mysql> SELECT * FROM Videos WHERE ke = `2Df5hBE`  AND end < DATE_SUB(NOW(), INTERVAL 3 DAY)
    -> ;
ERROR 1054 (42S22): Unknown column '2Df5hBE' in 'where clause'
```

moeiscool, 2017-02-15:
oops i wrapped the value wrong. Sorry.

gets oldest 10 events in your sql
```
SELECT * FROM Videos WHERE ke="2Df5hBE" ORDER BY `time` ASC LIMIT 10
````

gets videos older than 3 days
```
SELECT * FROM Videos WHERE ke = "2Df5hBE"  AND end < DATE_SUB(NOW(), INTERVAL 3 DAY)
```

Bonnee, 2017-02-15:
```
mysql> SELECT * FROM Videos WHERE ke="2Df5hBE" ORDER BY `time` ASC LIMIT 10
    -> ;
+------------+---------+------+---------------------+----------+-----------+--------+---------------------+--------+
| mid        | ke      | ext  | time                | duration | size      | frames | end                 | status |
+------------+---------+------+---------------------+----------+-----------+--------+---------------------+--------+
| fYOIo3lZcP | 2Df5hBE | mp4  | 2017-02-15 18:32:05 |     NULL |  15354700 |      0 | 2017-02-15 18:45:00 |      1 |
| kSJfWOlIvq | 2Df5hBE | mp4  | 2017-02-15 18:32:05 |     NULL |  16735900 |      0 | 2017-02-15 18:45:00 |      1 |
| WFtPXEN90r | 2Df5hBE | mp4  | 2017-02-15 18:32:05 |     NULL |  17322900 |      0 | 2017-02-15 18:45:00 |      1 |
| Azj1oq0nwB | 2Df5hBE | mp4  | 2017-02-15 18:32:05 |     NULL |  23764600 |      0 | 2017-02-15 18:45:00 |      1 |
| C9YopgTU8c | 2Df5hBE | mp4  | 2017-02-15 18:32:05 |     NULL |  67172200 |      0 | 2017-02-15 18:45:00 |      1 |
| WJWEpQiG8a | 2Df5hBE | mp4  | 2017-02-15 18:32:06 |     NULL |  67620100 |      0 | 2017-02-15 18:45:01 |      1 |
| c6QRUTBLJ6 | 2Df5hBE | mp4  | 2017-02-15 18:32:06 |     NULL | 194739000 |      0 | 2017-02-15 18:45:01 |      1 |
| kQaAUTqW4S | 2Df5hBE | mp4  | 2017-02-15 18:32:06 |     NULL |  24455800 |      0 | 2017-02-15 18:45:01 |      1 |
| G29O02Imfe | 2Df5hBE | mp4  | 2017-02-15 18:32:06 |     NULL | 194863000 |      0 | 2017-02-15 18:45:01 |      1 |
| Qj51yOMYey | 2Df5hBE | mp4  | 2017-02-15 18:32:06 |     NULL |  63686700 |      0 | 2017-02-15 18:45:01 |      1 |
+------------+---------+------+---------------------+----------+-----------+--------+---------------------+--------+
10 rows in set (0.01 sec)

mysql> SELECT * FROM Videos WHERE ke = "2Df5hBE"  AND end < DATE_SUB(NOW(), INTERVAL 3 DAY)
    -> ;
Empty set (0.00 sec)
```

I think they work properly.

moeiscool, 2017-02-15:
yah it is, currently I can't get duration from webm files so i didn't use the column just yet. Did you set `utcOffset` in conf.json? 

Are those the 10 oldest events in your db? and you have 0 over 3 days old because the cron wiped it yah?

Bonnee, 2017-02-15:
Yes, I have set utcOffset and yes, these were my oldest events.

I don't know if solving this will help someone, but I have the possibility to reinstall Shinobi in order to solve this. Let me know what's the best to do

moeiscool, 2017-02-15:
I am fairly certain it behave the same way even if you reinstall. I find it strange because i am using that version of the cron right now and last night and it worked fine for me. It didn't seem to do the interval thing though not sure what that is about I am digging at it. It successfully purged all files that were too old or beyond my set storage limit (100GB).

Bonnee, 2017-02-15:
As far as I remember, the problem came up after I did the sql update. I had done it some days after I updated the code because i noticed problems (I used update.sh without checking on github).

moeiscool, 2017-02-15:
strange because the update was only for adding a table for Events and change a column's type to varchar.

Can you please try installing it again? and just to be sure its only the cron messing up. Other stuff is ok?

Bonnee, 2017-02-15:
I installed it some hours ago and got no luck...

Bonnee, 2017-02-16:
Of course nothing changed

moeiscool, 2017-02-16:
what OS? Please find me on Slack https://shinobicctv.herokuapp.com/ this is breaking my heart. Now you say this started happening recently? what commit were you on when it worked and what commit did you update to when it stopped working?

that is probably a lot to ask for but I am hoping :crossed_fingers:

moeiscool, 2017-02-17:
still getting this?

Bonnee, 2017-02-17:
Nope! I downloaded cron.js again and that semms to have fixed it!

Thanks!

Bonnee, 2017-02-17:
Ok, nevermind. It took a while to happen but cron has deleted everything just now.

that's camera-out-0.log
```
CRON :  { f: 'did',
  msg: '2226 SQL rows with no file deleted',
  ke: '2Df5hBE',
  time: '2017-02-17T10:12:40.074Z' }
CRON :  { f: 'did',
  msg: '2242 SQL rows with no file deleted',
  ke: '2Df5hBE',
  time: '2017-02-17T10:13:23.416Z' }
CRON :  { f: 'did',
  msg: '10 old events deleted because over max of 100000 MB',
  ke: '2Df5hBE',
  time: '2017-02-17T10:13:25.278Z' }

```

moeiscool, 2017-02-18:
so i have added the options we talked about on Slack. It will allow you to turn off that part of the sequence (and others). I added definitions for the options in the config page https://github.com/moeiscool/Shinobi/wiki/Configuration

I also believe I may have fixed the issue. We're you using a custom video directory? the cron file was actually hardcoded to read `videos` folder inside the shinobi directory so it would likely say the files don't exist because it actually cant find them.

Until I confirm my suspicion to be true I will leave this thread open.

Bonnee, 2017-02-18:
Yea! I use a different directory for video, that's probably it!

moeiscool, 2017-02-18:
Can you please confirm this with the latest update. To see if it deletes everything everything? Then I'll feel comfortable closing this thread :D

Bonnee, 2017-02-19:
That's seems to work! Thanks!

closed 2017-02-19 by Bonnee