moeiscool/Shinobi #72: deleted files/events summary reappears on overlay on browser restart

clessley, 2017-02-16:
Running Shinobi within a x32 Ubuntu MATE via Virtual Box v4.3.12 on Windows 7.

If one deletes recorded events from the calendar they delete from the camera summary - blank list as expected but will appear in the main page overlay once browser is restarted.

The files themselves are being deleted from the /shinobi/videos/xxxx/xxxx/ folder.

moeiscool, 2017-02-16:
So the specified files are being removed but SQL rows are not?

you delete videos from calendar, files are removed but sql row is never removed? maybe you don't have sufficient privleges for this SQL user?

when you run `pm2 logs` in terminal, then do the delete command, what do you see in the terminal? any errors?

moeiscool, 2017-02-16:
what version are you running? that is the classic dashboard.

clessley, 2017-02-16:
The pm2 logs did nothing when deleting calendar events. No error or change.

Update.sh output a lot of permission errors (.npm, .pm2 directories) so I chmod -R 777 /shinobi and re-ran the update.sh again as root directly (not my root user). Rebooted, started pm2/mysql, opened firefox and the deleted entries were gone as desired from the summary panel. Camera is working.

Can you tell me what the proper permissions should be for the shinobi folder? Should they be owned by root or by a root user?

moeiscool, 2017-02-16:
I use root for everything even though I'm told it's not such a good idea. I do this because you have to be root to bind to any port under 1000. I use port 80 so I have to be root to use Shinobi. The directory itself can be owned by a non elevated user. It's probably better that way in case you decide to use a higher port.

note : that running `camera.js`, `cron.js`, `INSTALL.sh`, or `UPDATE.sh` as root will result in some directories and files to be owned by root. If you decide to run the app after as a regular user many things will not work.

since it is as expected I am just gonna hit that close button :smile: Thanks for trying Shinobi!

closed 2017-02-16 by moeiscool