moeiscool/Shinobi #73: Enhancement: Video record directory

Silverbeard00, 2017-02-17:
It'd be very cool if we could select which directory the video's save too.  I dunno if it's possible to also have it on some sort of max space usage too.

For instance being able to say it should record only 5Gb's worth and it would be able to delete old videos to record new ones.  Cyclical if you will.

moeiscool, 2017-02-17:
What version are you running? Some older versions do not have directory selection or max storage space.

To set the video directory you can edit your conf.json and add `videosDir`. More info about it can be found in the wiki.
https://github.com/moeiscool/Shinobi/wiki/Configuration

as for max space you can set the max space in your account settings. Right now the max space box shows for sub accounts but it does nothing. So you must be logged in to the main account to set it.

Silverbeard00, 2017-02-17:
Oh I see it now.  Absolutely perfect.

I literally spent all day yesterday trying to get zoneminder to work and it turned out to be a massive piece of crap.

Git cloned Shinobi at the end of the day and it was working in like 5 minutes.  Seems to have everything I need to!  Going to be using this for my business.

Will try to donate or contribute to help out cause this is fantastic.

closed 2017-02-17 by moeiscool