moeiscool/Shinobi #75: Monitor thumbnails erroring

Bonnee, 2017-02-19:
Since when I first installed Shinobi I had no thumbnails working on the dash, only 3 white dots on red. No big deal for me, so I never argued with that. Since the last 1-2 days, I get errors logged and the thumbnails on the dash are all messed up with an error writing or with the classic "no image" badge. Here's a screenshot
[image]

camera-error-0.log gets spammed by this:
```
uncaughtException TypeError: "listener" argument must be a function
    at _addListener (events.js:217:11)
    at Socket.addListener (events.js:277:10)
    at Socket.Readable.on (_stream_readable.js:687:35)
    at e.draw (/home/bonnee/Shinobi/camera.js:766:70)
    at Socket.<anonymous> (/home/bonnee/Shinobi/node_modules/connection-tester/index.js:53:16)
    at emitNone (events.js:86:13)
    at Socket.emit (events.js:186:7)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1077:10)
```

And in the dashboard console I see this:
[image]

Here's a closeup of some lines:
[image]

Here's the dashboard log output for one of the monitors:

```
2Df5hBE : mFDaFq7mxU
Snapshot Error a minute ago
cmd : ffmpeg -loglevel quiet -i "rtsp://undefined:undefined@192.168.1.126:554/live.sdp" -r 25 -ss 1.8 -frames:v 1 -f singlejpeg pipe:1 -y
```

The streams and the recordings are working fine though.
None of the stream types seems to work.

moeiscool, 2017-02-19:
Yah, thats in the dev branch. Newest dev branch commit will have it working. :D almost got v1 for motion detection, gonna sync it at the same time.

Bonnee, 2017-02-19:
Awesome news! Thanks!

Bonnee, 2017-02-23:
I saw you committed a fix, but I still have this issue. Is this normal?

moeiscool, 2017-02-23:
you still get Error in your preview icon? weird because I removed the line of code that says error.

you're still getting weird data back for the snapshot?

Bonnee, 2017-02-23:
Yes, I'm getting the exact same errors as before.

moeiscool, 2017-02-23:
does it say undefined for user and pass still? i see that you are on slack. wanna join me there? all these posts are gonna blast people's emails :P

closed 2017-03-04 by moeiscool