moeiscool/Shinobi #76: [Dev branch] Max storage amount not saving

Bonnee, 2017-02-23:
When I try to set the max storage amount, to 950000 for example, it just gets ignored and when I restart Shinobi, I get this:
```
camera-out-0.json

CRON :  { f: 'did',
  msg: '10 old videos deleted because over max of 10000 MB',
  ke: '2Df5hBE',
  time: '2017-02-20T20:42:30.424Z' }

```
And the older videos get deleted.

moeiscool, 2017-02-23:
i just tested saving it to my db and it did it correctly. Can you save any other fields? does 950000 show up as the limit when you refresh and open settings again?

does it seem to simply ignore your setting? or does it not get set in SQL at all?

Bonnee, 2017-02-23:
If I set the value and then refresh, it shows up 10000 again, same problem with max days to keep videos. I suppose that the db doesn't get set, but apart from this the db seems to be working fine.

moeiscool, 2017-02-26:
forgot to close this. issue was fixed in #78

closed 2017-02-26 by moeiscool