moeiscool/Shinobi #77: vagrant setup on os x

SamDecrock, 2017-02-24:
Hi, 

I'm trying this out for the first time using the vagrant instructions.

However, there's noting running port 8080. Even if I log into the VM with `vagrant ssh` and check with `curl http://localhost:8080` there's nothing running.

Is there somewhere I can look for log files to see what's going on?

Thanks,
Sam

moeiscool, 2017-02-24:
I just contacted @kzap, hopefully he will chime in. As he was the one who made the vagrant files. Personally I know nothing about vagrant but I will do my best.

I found this video on installing vagrant... but I am sure you already know how to install it. https://www.youtube.com/watch?v=RhhF8Yh7OnE . Hopefully there is a hint somewhere in there.

Can you run the installation with it on another port? Perhaps I can just update the ova image? Would that be acceptable?.. but of course it doesn't deal with the immediate issue... it'll just get your test machine going.

SamDecrock, 2017-02-24:
It's OK, I'll just try the manual installation instead. Just wanted a quick look at the applications ;-)

moeiscool, 2017-03-04:
I couldn't get a response from kzap so I have removed vagrant files. Sorry to anyone who couldn't get it going.

closed 2017-03-04 by moeiscool