moeiscool/Shinobi #80: Possibility of LDAP support

itisNathaniel, 2017-03-05:
Wow, this project is awesome, keep up the great work!
After spending many hours in the past then giving up with Zoneminder, it's awesome how well this 'just works'. I've only been playing for 5 minutes so far.

Is there any way to set up LDAP authentication out of curiosity, so I can link it to my home identity server (Synology LDAP) as my family struggles to remember more than one password!

I hope to be able to contribute back at some point too, but in the mean time, awesome job guys.

moeiscool, 2017-03-05:
Hehe, thanks. This was asked previously but I don't understand how it works. When I search ldap node.js I get these weird server things. Any tuts or sample code would be cool.

itisNathaniel, 2017-03-05:
Must say that I'm not a Node Dev myself (_yet!_) but maybe [this](https://github.com/trentm/node-ldapauth), [this](https://github.com/vesse/passport-ldapauth) or [this](https://www.npmjs.com/package/LDAP) module could help?
In terms of LDAP itself works, it's basically an open authentication protocol (opposed to something proprietary like Microsoft's Active Directory) that allows users to enter their credentials and it's checked against the server and access is allowed/denied. 

Also I've had a bit of an issue trying to set up a HTTPS for it, I set up proxy with apache for the app, going from http://192.168.1.11:8080 to https://code.domain.com. Using the proxy settings 

```
<IfModule mod_ssl.c>
<VirtualHost *:443>
    ServerName cctv.domain.uk
    ProxyPreserveHost On
    SSLProxyEngine On

    SSLEngine on
    SSLCertificateFile /cert.crt
    SSLCertificateKeyFile /cert.pem

ProxyPass / http://192.168.1.11:8080/
ProxyPassReverse / http://192.168.1.11:8080/
ProxyRequests off
</VirtualHost>
</IfModule>

```
And login still happens but the page loses all elements and I get this in the log http://imgur.com/a/Vp58y
```
0|camera   |   code: 'ER_DBACCESS_DENIED_ERROR',
0|camera   |   errno: 1044,
0|camera   |   sqlState: '42000',
0|camera   |   fatal: true }
0|camera   | Error Connecting : DB { [Error: ER_DBACCESS_DENIED_ERROR: Access denied for user 'shinobi'@'%' to database 'ccio']
0|camera   |   code: 'ER_DBACCESS_DENIED_ERROR',
0|camera   |   errno: 1044,
0|camera   |   sqlState: '42000',
0|camera   |   fatal: true }
0|camera   | Shinobi - PORT : 8080, NODE.JS : v4.2.6
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v4.2.6
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v4.2.6
0|camera   | 
```

Any ideas? Thanks

moeiscool, 2017-03-05:
When I use SSL I just put it on port 80 and use cloudflare, gotta use a domain though. The websocket part works with their caching engine... also instead of 8080 you should try 8443 of you want to use ssl without the standard ports.

update that node.js! 7.2.0+ required https://davidwalsh.name/upgrade-nodejs

Also it seems your conf.json is misconfigured or you have not added the sql/user.sql file to your DB.

itisNathaniel, 2017-03-05:
Cheers, I think that node outdated version caused me my install problems 😂. And logging into mysql and importing user.sql seems to get rid of that message.
Thanks, will try setting up cloudflare for it.

moeiscool, 2017-03-05:
im using socket.io+expressjs. Their site is down right now so i can't link you to their example but this one is just as good, if not better. http://www.programwitherik.com/socket-io-tutorial-with-node-js-and-express/

websocket and webserver run on the same port, default is 8080. If this blows your mind, dw. It blew my mind ten times over.

moeiscool, 2017-03-05:
also as this original thread started as a suggestion I'll be adding it to slack suggestions channel. Please post future suggestions there as it also gives you a chance to see if your suggestions has already been suggested :smile: 

Here is the page to get an invite : https://shinobicctv.herokuapp.com/

closed 2017-03-05 by moeiscool