moeiscool/Shinobi #86: Add to Docker Hub and set up Automated Build

tfohlmeister, 2017-03-09:
Would you please add this to Docker Hub and create an Automated Build?
This way it's way easier to use in any Docker environment. I cloned the repo and set it up under my account so you can check it out: https://hub.docker.com/r/tfohlmeister/shinobi/

moeiscool, 2017-03-09:
Sorry I don't follow. This website will make a docker file for me? Personally I don't user docker so I am inept when it comes to it.

Or are there files in your repo that I can add to master?

tfohlmeister, 2017-03-09:
I see no worries!
So the docker build file someone added to this repo allows you to build a docker image manually (and locally). Docker Hub is a free registry for storing docker images. They can also build these images for you, if you link a GitHub repo. Then every time you push changes to this repo a built is automatically triggered which is really cool.
Anyway you have to be the owner of the repository at GitHub to setup up Docker Hub automatic builds. But no worries, I can live with a cloned repo for now.

closed 2017-03-09 by tfohlmeister

moeiscool, 2017-04-28:
#120 Docker now works. Info here.