moeiscool/Shinobi #88: a couple of questions - autostart & app

aristosv, 2017-03-13:
First of all thank you for creating an alternative to the existing dinosaurs. Hopefully this will bring about some healthy antagonism. I've been using Shinobi for less than an hour now, and I must say I'm already impressed with the simplicity of the installation, configuration and usage.

It's currently configured to pick up the stream from 4 Raspberry Pi's with NoIR cams, I've had zero difficulties configuring it and so far its doing a great job.

I would like to know if there's an iPhone app I can use to view the camera feeds, and how would you configure Ubuntu to auto-start Shinobi.

Thanks.

moeiscool, 2017-03-13:
yay 😄ill be adding it in the docs soon. with pm2 you can run `pm2 startup` then `pm2 save` after starting shinobi. it will then start on boot 😄

as for an actual iphone app, no. but saving the app to home screen makes it appear as one without the URL bar 😄

closed 2017-03-13 by aristosv

Timo-Linde, 2017-03-17:
+1 for adding in the docs.
I will install Shinobi for a 20 Camera System on a 
Dell PowerEdge R710 with 
2x Intel(R) Xeon(R) CPU X5670 @ 2.93GHz => 24 Logical Cores
96GB Ram
and 12 TB Raid 5