moeiscool/Shinobi #89: Docker cleanup?

jacobalberty, 2017-03-14:
I'm happy to submit the pull request for it but I wanted to make a few suggestions to clean up the Dockerfile and get your input before doing so in case theres a reason you did things the way you did

- docker-entrypoint.sh modifies the configuration, this should probably be done in Dockerfile instead so it's only done once and not every time the container gets run 
- Export port 8080 in the Dockerfile just to fit normal standards.
- Include some documentation for how the docker container is to be used. 

The documentation should probably go in README.md so if you ever add this to the hub it will show up there, but I'm shit for writing documentation so in my PR i'll probably make a seperate DOCKER.md that you can later merge into README.md if you so desire.

moeiscool, 2017-03-14:
Please do! The docker stuff was made by another user so I don't know it's integrity. All I know is they can be used 😄. As for the readme you can throw it in the INSTALL folder. 

Actually it would nice if you could put all the docker files into a docker folder inside INSTALL. I didn't because I wasn't sure if it would break the files or not. But since you know what you're doing with them : maybe you could? Just a suggestion its not necessary... but the DOCKER.md should go in INSTALL and I'll add it to the docs later.

Thank you! 😁 Side note : please post suggestions and questions on Reddit or slack. My OCD wants to use the github issues only for issues 😅 Thank you again!

closed 2017-03-14 by moeiscool

moeiscool, 2017-03-14:
also a side note please do pull requests to the dev branch. https://github.com/moeiscool/Shinobi/tree/dev

moeiscool, 2017-04-28:
#120 Docker now works. Info here.