moeiscool/Shinobi #95: Unexpected token ...

Alko89, 2017-03-17:
I just went through the harder way installation on Ubuntu and am unable to execute `nodejs camera.js`.
The only error I get is `uncaughtException [SyntaxError: Unexpected token ...]`. 
I have installed Shinobi in a virtual machine a few weeks ago with no problem.

Timo-Linde, 2017-03-17:
i had the same problem on debian over here i just installed a newer nodejs version

https://nodejs.org/en/download/package-manager/

Alko89, 2017-03-17:
Thanks! That solved the issue. Strange it seems it works on Ubuntu LTS.

clessley, 2017-03-17:
No its doesn't, got the same error using it on ubuntu yesterday after upgrade script and clean install.

Timo-Linde, 2017-03-17:
@clessley Can you please provide `node -v`

moeiscool, 2017-03-17:
nodejs 7.2.0(+) is required to run Shinobi.

you can update node.js like this if you dont have nvm :
```
sudo npm cache clean -f
sudo npm install -g n
sudo n stable
```

closed 2017-03-17 by moeiscool

Alko89, 2017-03-17:
I admit, I don't remember what I did that it worked previously. Maybe I did updated node through npm.
Anyway its working now. Thank you.

moeiscool, 2017-03-17:
it worked before because the older version of node that you had was capable of running the version of `camera.js` that you had. As low as 4.x would work on older versions. Now it has methods and functions that the older versions are no longer capable of running.

clessley, 2017-03-21:
@Timo-Linde This was closed before I had a chance to retry the latest version. The latest on Ubuntu MATE 16.04 LTS inside a virtual box on Windows 7 installs ok and the previous error has been fixed.