moeiscool/Shinobi #98: stops recording after deleting segment

Timo-Linde, 2017-03-20:
Shinobi stops recording after deleting one or more segments over the UI from the past. e.g. one hour ago.

Shinobi records the actual segment to the end and then stops.

The UI gives me this error: Cannot save non existant file. Something went wrong.

moeiscool, 2017-03-20:
Any logs in pm2? Are clocks adjusted for daylight savings? I was having timestamp issue a week ago, I thought I fixed it. Guess not. What OS did you install on? When you refresh does the GUI appear normal?

Deleting any video will halt recording? Or jus the last one?

Timo-Linde, 2017-03-21:
no logs in pm2.  
OS:  Debian GNU/Linux 8 (64 Bit)

The video remains in the list till i reopen the list.
Deleting any video will halt recording after the segment is finished.

I refreshed the GUI and it appears normal.

moeiscool, 2017-03-21:
any logs in your browser console when running the command? could you please paste the last few outputs after running the command.

closed 2017-04-04 by moeiscool