moeiscool/Shinobi #99: UI timezone not adapted

Timo-Linde, 2017-03-20:
[image]

Bonnee, 2017-03-20:
I have a similar issue: The recordings' file name is right with my timezone, but the one reported by the UI doesn't consider the offset: it reports a time with 0 offset (e.g. 2017-03-20T15:45:01.000Z while the filename is 2017-03-20T16-45-01.mp4). Querying the db, the time seems fine. So it's definitely an UI issue. The clock in the dash is right tho.

I too have the offset set to +0100

moeiscool, 2017-03-20:
hmm i thought i fixed the timestamp thing, guess not. No errors appear in `pm2 logs`? is daylight savings correctly set on the machine you have shinobi installed?

Bonnee, 2017-03-20:
Yes, I have DST set correctly and no, no errors pop up in the logs

Timo-Linde, 2017-03-21:
No errors in `pm2 logs`
OS: Debian GNU/Linux 8 (64 Bit)
Daylight saving time seems to work properly
[image]

moeiscool, 2017-03-21:
Any errors in your browser console when trying to delete? Can you paste the last few console outputs when doing the delete please.

Bonnee, 2017-03-21:
I get no output at all when deleting

moeiscool, 2017-03-21:
i mean in browser console on the client side. Not terminal.

Bonnee, 2017-03-21:
Yes, I get no errors from anywhere. In fact I never noticed this bug until I came across this issue

moeiscool, 2017-03-22:
what outputs do you get? can you paste those objects here? it should return `video_delete` in one of them

Timo-Linde, 2017-03-22:
[image]

moeiscool, 2017-03-24:
sorry i havent got back to this issue. i have been working on a few things in motion. ah yah not video_delete but will look into it that output is helpful.

moeiscool, 2017-04-11:
i think i fixed this and i haven't run into this issue for a while now. going to close unless you are experiencing this on the latest master or dev.

closed 2017-04-11 by moeiscool

Bonnee, 2017-04-12:
I still have this issue, and after the daylight change the delay has increased to 2 hours

moeiscool, 2017-04-12:
is your server's time and sql time correct? I actually experienced this yesterday and the fix was correcting the offset in `my.cnf` for sql.
I actually experienced this yesterday and the fix was correcting the offset in `my.cnf` for sql.

moeiscool, 2017-04-20:
newest commit fixes this