moeiscool/Shinobi #101: cannot login after fresh install?

clessley, 2017-03-21:
logging into shinobi at localhost:8080 fails, the page blinks but does not login to camera control

I am using `ccio@m03.ca` and `password`. Using my own email/pw via config also exhibits the same behavior. In the past the former was the way to gain access to admin to set up person login credentials.

Ubuntu MATE 16.04 x86 via virtual box on Windows 7.

moeiscool, 2017-03-21:
did you set up `conf.json` with your SQL details? Please check `pm2 logs` to see if there are any errors.

Did you add the file `sql/default_data.sql` in the sql table?

In the near future the install process will ask what credentials you want to use.

clessley, 2017-03-21:
Conf.json is set up with SQL credentials yes.

I have one row in users, majesticflame as expected and I have the ccio table too. The user.sql import fails, so I manually added a user via adminer but same blinking login after restart. 

Here are pm2 logs -err:

[image]

Perhaps I should purge and reinstall fresh, again? It was so easy the first time.

clessley, 2017-03-21:
Purged mysql server, purge shinobi folder, re-install "the ubuntu easy way" and then did the all of the "ubuntu harder way" sql commands, restart mysql, delete exiting conf.json, rename conf.example.json to conf.json = login ok with Bunny cams.

moeiscool, 2017-03-21:
it looks like you are missing dependencies. hmmmm. I guess the installer didn't work correctly for your system.

The solution would have been to run `npm install` sorry about that.

closed 2017-03-21 by moeiscool

clessley, 2017-03-21:
The first time i downloaded from guthub clone via browser, the second time strictly through the command line.
-- 
Sent from my mobile device. Please excuse my brevity.