moeiscool/Shinobi #102: Cron not deleting files

Bonnee, 2017-03-25:
I have set the "Number of Days to keep Videos" to 3, and that actually worked fine. I then discovered than the disk space for only 3 days of recordings was a bit high, so I checked the actual direcotries of the monitors, and I found very old recordings sitting there.
I found no strange messages or errors in the logs

Bonnee, 2017-03-26:
I've just found that even if the time to keep videos gets increased, older than 3 days recordings disappear.
Looking at the logs, these error have come out:
```
Error: Connection lost: The server closed the connection.
    at Protocol.end (/opt/Shinobi/node_modules/mysql/lib/protocol/Protocol.js:109:13)
    at Socket.<anonymous> (/opt/Shinobi/node_modules/mysql/lib/Connection.js:109:28)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:188:7)
    at endReadableNT (_stream_readable.js:975:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickDomainCallback [as _tickCallback] (internal/process/next_tick.js:128:9)
uncaughtException { Error: Connection lost: The server closed the connection.
    at Protocol.end (/opt/Shinobi/node_modules/mysql/lib/protocol/Protocol.js:109:13)
    at Socket.<anonymous> (/opt/Shinobi/node_modules/mysql/lib/Connection.js:109:28)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:188:7)
    at endReadableNT (_stream_readable.js:975:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickDomainCallback [as _tickCallback] (internal/process/next_tick.js:128:9) fatal: true, code: 'PROTOCOL_CONNECTION_LOST' }
```
But I don't know If they're related to the problem

moeiscool, 2017-03-26:
The error means SQL lost connection. I was looking into this yesterday, got orphaned files dealt with it. Going to rewrite other parts. It was working, I'm not sure what's I did for it to make it start deleting all files again.

Bonnee, 2017-03-26:
Great to see that you're already on it!

moeiscool, 2017-03-28:
after many tests my files are still lingering! im going to close this.

closed 2017-03-28 by moeiscool