moeiscool/Shinobi #109: Mac OS install failure

YoureNowOnTV, 2017-04-10:
G'day there,

I was running through the steps listed for the Mac OS install for Shinobi and have run into an issue on the entry line:

cat sql/framework.sql | mysql -u root -p

I get the error message:

-bash: mysql: command not found

I'm not sure what I've missed that Terminal can't find mysql ?

Cheers.

Bonnee, 2017-04-10:
Have you installed mysql? Is it in your $PATH? What happens if you do 
```
which mysql
```
in the terminal?

YoureNowOnTV, 2017-04-10:
Thanks for the reply !

I've installed MySql as far as I understand ?

If I type:

`which mysql`

It just goes to the next terminal line.

moeiscool, 2017-04-10:
if you have mysql installed and can connect to it via a client, like HeidiSQL or command line, then you can import `framework.sql`, `default_data.sql` and `user.sql` into sql manually.

YoureNowOnTV, 2017-04-10:
Thanks for assisting moeiscool, I'm a bit lost here.

I've played with sql on a web server but never locally so I'm flying blind in what to do, to get it running, so I can go to the next step to get Shinobi running.

Sorry to be a newb !

moeiscool, 2017-04-10:
if you have phpmyadmin or something similar installed you can use that to import the sql files. Otherwise i suggest trying to setup in ubuntu under virtualbox. Then attempting mac, just so you can get a feel of what is going on with the install process.

closed 2017-04-11 by moeiscool