moeiscool/Shinobi #114: Raspberry Pi cheap Sricam no stream (errors in P frame)

codingyourlife, 2017-04-13:
I have a thumbnail that seems up2date but I cannot receive the stream. On Windows I have ONVIF Device Manager which can receive the stream so I know Ip, Port, Path setting is correct (not sure about the rest).

I'm on a Raspberry Pi and installed ffmpeg like this:
```
wget https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_3.1.1-1_armhf.deb
dpkg -i ffmpeg_3.1.1-1_armhf.deb
```

My camera is: Scricam SP013 (was hard to find out, hope it's correct)
Stream is under: rtsp://[local-ip]:554/onvif1

Here my probe result:
```
ffprobe version 3.1.1
Copyright (c) 2007-2016 the FFmpeg developers

built with gcc 4.9.2 (Raspbian 4.9.2-10) 
configuration: --prefix=/usr --enable-gpl --enable-libx264 --enable-nonfree --enable-shared --disable-static 
libavutil 55. 28.100 / 55. 28.100 libavcodec 57. 48.101 / 57. 48.101 libavformat 57. 41.100 / 57. 41.100 libavdevice 57. 0.101 / 57. 0.101 libavfilter 6. 47.100 / 6. 47.100 libswscale 4. 1.100 / 4. 1.100 
libswresample 2. 1.100 / 2. 1.100 libpostproc 54. 0.100 / 54. 0.100 
[h264 @ 0xce58e0] corrupted macroblock 15 1 (total_coeff=16) [h264 @ 0xce58e0] error while decoding MB 15 1 [h264 @ 0xce58e0] concealing 3554 DC, 3554 AC, 3554 MV errors in P frame 
[h264 @ 0xce58e0] 
Increasing reorder buffer to 1 
[h264 @ 0xce58e0] 
top block unavailable for requested intra mode 
[h264 @ 0xce58e0] 
error while decoding MB 69 0 [h264 @ 0xce58e0] concealing 3580 DC, 3580 AC, 3580 MV errors in P frame 
[h264 @ 0xce58e0] 
top block unavailable for requested intra mode -1 
[h264 @ 0xce58e0] 
error while decoding MB 69 0 
[h264 @ 0xce58e0] 
concealing 3580 DC, 3580 AC, 3580 MV errors in P frame 
[h264 @ 0xce58e0] 
out of range intra chroma pred mode 
[h264 @ 0xce58e0] 
error while decoding MB 60 1 
[h264 @ 0xce58e0] 
concealing 3509 DC, 3509 AC, 3509 MV errors in P frame 
[h264 @ 0xce58e0] negative number of zero coeffs at 63 0 [h264 @ 0xce58e0] error while decoding MB 63 0 
[h264 @ 0xce58e0] concealing 3586 DC, 3586 AC, 3586 MV errors in P frame 
[h264 @ 0xce58e0] mb_type 29 in P slice too large at 66 0 [h264 @ 0xce58e0] error while decoding MB 66 0 
[h264 @ 0xce58e0] concealing 3583 DC, 3583 AC, 3583 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode -1 [h264 @ 0xce58e0] error while decoding MB 72 0 
[h264 @ 0xce58e0] concealing 3577 DC, 3577 AC, 3577 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode [h264 @ 0xce58e0] error while decoding MB 65 0 
[h264 @ 0xce58e0] concealing 3584 DC, 3584 AC, 3584 MV errors in P frame 
[h264 @ 0xce58e0] 
top block unavailable for requested intra mode 
[h264 @ 0xce58e0] error while decoding MB 66 0 
[h264 @ 0xce58e0] concealing 3583 DC, 3583 AC, 3583 MV errors in P frame 
[h264 @ 0xce58e0] P sub_mb_type 5 out of range at 64 0 [h264 @ 0xce58e0] error while decoding MB 64 0 
[h264 @ 0xce58e0] concealing 3585 DC, 3585 AC, 3585 MV errors in P frame 
[h264 @ 0xce58e0] mb_type 26 in P slice too large at 66 0 [h264 @ 0xce58e0] error while decoding MB 66 0 
[h264 @ 0xce58e0] concealing 3583 DC, 3583 AC, 3583 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode [h264 @ 0xce58e0] error while decoding MB 65 0 [h264 @ 0xce58e0] concealing 3584 DC, 3584 AC, 3584 MV errors in P frame 
[h264 @ 0xce58e0] mb_type 110 in P slice too large at 13 1 [h264 @ 0xce58e0] error while decoding MB 13 1 
[h264 @ 0xce58e0] concealing 3556 DC, 3556 AC, 3556 MV errors in P frame 
[h264 @ 0xce58e0] 
top block unavailable for requested intra mode -1 
[h264 @ 0xce58e0] 
error while decoding MB 65 0 
[h264 @ 0xce58e0] 
concealing 3584 DC, 3584 AC, 3584 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode [h264 @ 0xce58e0] error while decoding MB 70 0 [h264 @ 0xce58e0] concealing 3579 DC, 3579 AC, 3579 MV errors in P frame 
[h264 @ 0xce58e0] out of range intra chroma pred mode [h264 @ 0xce58e0] error while decoding MB 71 0 
[h264 @ 0xce58e0] concealing 3578 DC, 3578 AC, 3578 MV errors in P frame 
[h264 @ 0xce58e0] negative number of zero coeffs at 41 1 [h264 @ 0xce58e0] error while decoding MB 41 1 
[h264 @ 0xce58e0] concealing 3528 DC, 3528 AC, 3528 MV errors in P frame 
[h264 @ 0xce58e0] out of range intra chroma pred mode [h264 @ 0xce58e0] error while decoding MB 13 1 
[... 87 lines cut ...]
[h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode [h264 @ 0xce58e0] error while decoding MB 39 0 
[h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] mb_type 53 in P slice too large at 28 1 [h264 @ 0xce58e0] error while decoding MB 28 1 
[h264 @ 0xce58e0] concealing 901 DC, 901 AC, 901 MV errors in P frame 
[h264 @ 0xce58e0] top block unavailable for requested intra mode -1 [h264 @ 0xce58e0] error while decoding MB 9 0 
[h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] out of range intra chroma pred mode [h264 @ 0xce58e0] error while decoding MB 1 2 [h264 @ 0xce58e0] concealing 888 DC, 888 AC, 888 MV errors in P frame 
[h264 @ 0xce58e0] left block unavailable for requested intra mode [h264 @ 0xce58e0] error while decoding MB 0 2 [h264 @ 0xce58e0] concealing 889 DC, 889 AC, 889 MV errors in P frame 
[h264 @ 0xce58e0] cbp too large (192) at 10 0 [h264 @ 0xce58e0] error while decoding MB 10 0 
[h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] cbp too large (413) at 19 2 [h264 @ 0xce58e0] error while decoding MB 19 2 
[h264 @ 0xce58e0] concealing 870 DC, 870 AC, 870 MV errors in P frame 
[h264 @ 0xce58e0] P sub_mb_type 32 out of range at 16 1 [h264 @ 0xce58e0] error while decoding MB 16 1 
[h264 @ 0xce58e0] concealing 913 DC, 913 AC, 913 MV errors in P frame 
[h264 @ 0xce58e0] mb_type 249 in P slice too large at 8 1 [h264 @ 0xce58e0] error while decoding MB 8 1 
[h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] dquant out of range (-127) at 22 0 [h264 @ 0xce58e0] error while decoding MB 22 0 [h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[h264 @ 0xce58e0] P sub_mb_type 6 out of range at 36 0 [h264 @ 0xce58e0] error while decoding MB 36 0 [h264 @ 0xce58e0] concealing 920 DC, 920 AC, 920 MV errors in P frame 
[rtsp @ 0xce2a10] decoding for stream 0 failed 
Input #0, rtsp, from 'rtsp://192.168.178.67:554/onvif1': Metadata: title : H.264 Video, RtspServer_0.0.0.2 Duration: N/A, start: 0.000000, bitrate: N/A 
Stream #0:0: Video: h264 (Baseline), yuv420p, 640x360, 90k tbr, 90k tbn, 180k tbc 
Stream #0:1: Audio: pcm_alaw, 8000 Hz, 1 channels, s16, 64 kb/s 
END
```

moeiscool, 2017-04-13:
There are 2 ways the thumbnail works. One is that it opens its own connection and gets a frame from the camera or it uses an already generated image from RAM. When it's doing the first way some cameras cannot get the stream because the camera has a connection limit reached even though the snapshot process was using it for a few seconds.

To remedy this on some cameras you can enable Snapshot (CGI-BIN). This will make snapshots get from RAM.

closed 2017-04-13 by moeiscool

codingyourlife, 2017-04-14:
Nice, that worked. Thanks!