moeiscool/Shinobi #120: Docker not working

matansagee, 2017-04-25:
Hi i'm using your docker image `https://hub.docker.com/r/shinobicctv/shinobi/`
and i get the following output
```
sed: can't read /opt/shinobi/conf.json: No such file or directory
sed: can't read /opt/shinobi/conf.json: No such file or directory
sed: can't read /opt/shinobi/conf.json: No such file or directory
module.js:328
    throw err;
    ^

Error: Cannot find module '/opt/shinobi/camera.js'
    at Function.Module._resolveFilename (module.js:326:15)
    at Function.Module._load (module.js:277:25)
    at Function.Module.runMain (module.js:442:10)
    at startup (node.js:136:18)
    at node.js:966:3
```
the content of `/opt/shinobi` is:
```
drwxr-xr-x 4 root root 4096 Apr 25 08:36 ./
drwxr-xr-x 6 root root 4096 Apr 25 08:33 ../
-rw-r--r-- 1 root root   26 Mar 14 21:25 .dockerignore
-rw-r--r-- 1 root root  295 Mar 14 21:25 Dockerfile
-rw-r--r-- 1 root root  561 Mar 14 21:25 docker-compose.yml
-rwxr-xr-x 1 root root  347 Mar 14 21:25 docker-entrypoint.sh*
-rw-r--r-- 1 root root  115 Mar 14 21:25 docker.env
drwxr-xr-x 2 root root 4096 Apr 25 08:33 node_modules/
```
is there a running docker for the cctv?

Thanks

moeiscool, 2017-04-25:
currently the docker installer is broken.. mainly because I don't know how to set it up. :sweat_smile: I'll probably be looking into fixing it within the next few days.

If you have any input on how I can make it work it would be appreciated.

moeiscool, 2017-04-28:
Docker is now working!
http://shinobi.video/docs/start#docker

[image]

closed 2017-04-28 by moeiscool