moeiscool/Shinobi #125: Docker Persistence

tfohlmeister, 2017-04-29:
Hi,
please provide some information about how to setup docker in a persistent way after fixing it in #120.
I added a volume to be mounted at `/opt/shinobi/videos` to save videos outside the container but it seems the configuration is overwritten every time I start the container.
* Login credentials are back to default
* Previously configured cameras are missing

My guess is the mysql-server data needs to be saved somehow as well?!

Thanks!

moeiscool, 2017-04-29:
Well that's not good. Ill see if I can figure a way to make it not delete everything lol. I imagined it would be the same on every restart as the files are the same every restart. Ill look into it in the next few weeks.

moeiscool, 2017-04-29:
so i just ran my docker again and it ran how i expected it to. My stuff was still there.

Are you running the latest docker image? or an older one?

tfohlmeister, 2017-04-29:
Yeah sure, if you don't destroy the container your data is still there and I can tell from the `docker-entrypoint.sh` that you skip installation if `installed.txt` is present. So I get what you are trying to achieve here.
But in the docker style you usually want to separate configurations and persistent data from your actual docker container using for example [volumes](https://docs.docker.com/engine/tutorials/dockervolumes/). The reason being that everything inside your container is considered to be volatile. If you `docker pull` a new version of a container for example all your local changes get overwritten by the original container content once you `docker run` it.

So most `Dockerfile`s define several volumes to hint for folders which should be mounted outside the container for persistence. For Shinobi my guess would be that at least `/var/lib/mysql` should become volumes for that matter maybe a folder `config` in `/opt/shinobi`. The latter of course requires that you move all user configuration to that folder.

I might try to work on a PR for this as I find it interesting.

moeiscool, 2017-04-30:
I recently found out how docker works and I may not go into more detail with it. The docker installer does work, you just need to do some tweaking to make it usable.

Personally I feel docker, for Shinobi, is only good for testing and sampling. The weird port mapping doesn't sit well with me.

Please feel free to update the docker stuff to meet standards, as for me I'm going to keep my focus on other aspects.

If you would like me to spend more time on docker please consider ordering a support package. Thx!

closed 2017-04-30 by moeiscool