moeiscool/Shinobi #127: SyntaxError in shinobi.init

intensite, 2017-05-02:
Hello,  
I discovered your excellent software this week and it works great.  I use it to embed the stream of one surveillance camera in a internal web page.

However, the chrome console is polluted by the following error message each time the page where I have embedded Shinobi is displayed.  The following is the embeded iFrame code:

`<iframe src="http://192.168.222.218:8080/yeGbxp16yLiDdEXbLpldB9fZ8I3jpE/embed/2Df5hBE/MUx0zKBEPD/jquery|fullscreen" width="450" height="252"></iframe>`

The error produced is:
```
VM1880:1 Uncaught SyntaxError: Unexpected token o in JSON at position 1
    at JSON.parse (<anonymous>)
    at Object.$.shinobi.init (jquery|fullscreen:44)
    at Socket.<anonymous> (jquery|fullscreen:86)
    at Socket.Emitter.emit (socket.io.js:7414)
    at Socket.onevent (socket.io.js:7130)
    at Socket.onpacket (socket.io.js:7088)
    at Manager.<anonymous> (socket.io.js:7520)
    at Manager.Emitter.emit (socket.io.js:7414)
    at Manager.ondecoded (socket.io.js:2823)
    at Decoder.<anonymous> (socket.io.js:7520)
```
 
[image]

This is not a blocking error, I can see the camera stream.  So it is not urgent.

moeiscool, 2017-05-02:
Erg. Could you also post your camera configuration please. Just hitting export and uploading the file should do.

intensite, 2017-05-02:
Here it is:
```
{
	"mode" : "start",
	"mid" : "MUx0zKBEPD",
	"name" : "Porte",
	"type" : "h264",
	"protocol" : "rtsp",
	"host" : "192.168.223.107",
	"port" : "554",
	"path" : "h264Preview_01_main",
	"ext" : "mp4",
	"fps" : "1",
	"width" : "640",
	"height" : "480",
	"details" : "{\"fatal_max\":\"\",\"rtsp_transport\":\"no\",\"muser\":\"erp\",\"mpass\":\"password_removed\",\"sfps\":\"1\",\"aduration\":\"\",\"stream_type\":\"b64\",\"stream_mjpeg_clients\":\"\",\"stream_vcodec\":\"no\",\"stream_acodec\":\"no\",\"hls_time\":\"2\",\"preset_stream\":\"ultrafast\",\"hls_list_size\":\"3\",\"signal_check\":\"10\",\"signal_check_log\":\"0\",\"stream_quality\":\"10\",\"stream_fps\":\"5\",\"stream_scale_x\":\"640\",\"stream_scale_y\":\"480\",\"svf\":\"\",\"vcodec\":\"libx264\",\"crf\":\"1\",\"preset_record\":\"\",\"acodec\":\"none\",\"dqf\":\"0\",\"cutoff\":\"15\",\"vf\":\"\",\"timestamp\":\"0\",\"timestamp_font\":\"\",\"timestamp_font_size\":\"10\",\"timestamp_color\":\"white\",\"timestamp_box_color\":\"0x00000000@1\",\"timestamp_x\":\"(w-tw)/2\",\"timestamp_y\":\"0\",\"cust_input\":\"\",\"cust_snap\":\"\",\"cust_detect\":\"\",\"cust_stream\":\"\",\"cust_record\":\"\",\"custom_output\":\"\",\"snap\":null,\"snap_fps\":\"\",\"snap_scale_x\":\"\",\"snap_scale_y\":\"\",\"detector\":\"0\",\"detector_trigger\":null,\"detector_timeout\":\"\",\"watchdog_reset\":null,\"detector_save\":null,\"detector_mail\":\"0\",\"detector_mail_timeout\":\"\",\"detector_fps\":\"\",\"detector_scale_x\":\"\",\"detector_scale_y\":\"\",\"detector_frame\":\"1\",\"detector_sensitivity\":\"\",\"cords\":\"\",\"detector_face\":null,\"detector_fullbody\":null,\"detector_car\":null,\"detector_notrigger\":null,\"detector_notrigger_mail\":null,\"detector_notrigger_timeout\":\"\",\"control\":\"0\",\"control_base_url\":\"\",\"control_stop\":\"0\",\"control_url_stop_timeout\":\"\",\"control_url_center\":\"\",\"control_url_left\":\"\",\"control_url_left_stop\":\"\",\"control_url_right\":\"\",\"control_url_right_stop\":\"\",\"control_url_up\":\"\",\"control_url_up_stop\":\"\",\"control_url_down\":\"\",\"control_url_down_stop\":\"\",\"groups\":\"[]\",\"loglevel\":\"error\",\"sqllog\":\"0\"}",
	"shto" : "[]",
	"shfr" : "[]"
}

```

moeiscool, 2017-05-02:
i have been unable to reproduce this error, can you confirm this is happening on the dev branch as well?

closed 2017-05-04 by moeiscool