moeiscool/Shinobi #135: FFprobe does nothing

Whytehorse, 2017-05-12:
I clicked on FFprobe and entered
`rtsp://192.168.1.198:8554/`
Then I hit check and it does nothing. On the commandline I can run 
`ffprobe rtsp://192.168.1.198:8554/`
and I get
`   Last message repeated 1 times
[h264 @ 0x247f480] decode_slice_header error
[h264 @ 0x247f480] no frame!
Invalid UE golomb code
    Last message repeated 1 times
Input #0, rtsp, from 'rtsp://192.168.1.198:8554/':
  Metadata:
    title           : Unnamed
    comment         : N/A
  Duration: N/A, start: 596802.819333, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1280x720, 30 tbr, 90k tbn, 180k tbc
`

moeiscool, 2017-05-12:
hm. :smile: well i see success. I can give you a quick breakdown of whats going on here.

`Last message repeated 1 times [h264 @ 0x247f480] decode_slice_header error [h264 @ 0x247f480] no frame!`
probably intermittent connection to the location, maybe wifi or bad wiring.. it could be a number of things.

` Invalid UE golomb code Last message repeated 1 times`
beats the heck out of me.

` Input #0, rtsp, from 'rtsp://192.168.1.198:8554/': Metadata: title : Unnamed comment : N/A Duration: N/A, start: 596802.819333, bitrate: N/A Stream #0:0: Video: h264 (High), yuv420p, 1280x720, 30 tbr, 90k tbn, 180k tbc`

You have found a stream with 1 input.
Type : H.264
Pixel Format : yuv420p (this isn't an option in shinobi yet)
Width : 1280
Height : 720
FPS : 30, it says tbr because ffmpeg could not get the `fps` property and guessed it based on the sample stream it took.

The rest i have no options for but there is a description of them here http://stackoverflow.com/questions/3199489/meaning-of-ffmpeg-output-tbc-tbn-tbr
```
tbn = the time base in AVStream that has come from the container
tbc = the time base in AVCodecContext for the codec used for a particular stream
```

Whytehorse, 2017-05-12:
I've isolated this issue more and submitted a new bug https://github.com/moeiscool/Shinobi/issues/137

moeiscool, 2017-05-12:
oh sorry :( i missed half your post. The prober doesn't work?

can you run `ps aux | grep ffprobe` right after hitting check and see if it appears?

Whytehorse, 2017-05-12:
you can close this, ffprobe works when not having the login/logout issue

closed 2017-05-12 by moeiscool