moeiscool/Shinobi #136: Add raspberry pi to tested cameras

Whytehorse, 2017-05-12:
IT WORKS!!!! It's ALIVE!!!! Muahahaha. Tested raspberry pi camera. The rpi command to send video is 
`LD_PRELOAD='' cvlc v4l2:///dev/video0 --v4l2-width 1296 --v4l2-height 976 --v4l2-chroma h264 --sout '#rtp{sdp=rtsp://:8554/}' `

And the settings on shinobi are:
[image]
[image]

moeiscool, 2017-05-12:
woah nelly. If you could please hit Export then post the file here that would be sweet. Then i can make references here so it also gives credit to the tester :smile:

Whytehorse, 2017-05-12:
I can't upload a json file so it's tarred and gzipped.
[Shinobi_CZOfp5pwqb_config.json.tar.gz](https://github.com/moeiscool/Shinobi/files/994883/Shinobi_CZOfp5pwqb_config.json.tar.gz)

closed 2017-05-12 by moeiscool