moeiscool/Shinobi #137: Logout/login breaks shinobi

Whytehorse, 2017-05-12:
If I logout and then login I get a broken shinobi. I can fix it by running at the terminal:
`pm2 restart camera.js`
So it appears to be some kind of session or browser cache issue. If I do it quickly then sometimes it's not broken. I'm using chrome on ubuntu 16.04, connecting to localhost. If I logout and then login as admin, then logout, and login as my user, it appears as follows(missing cameras, ffprobe broken, missing cpu data, etc).
[image]

moeiscool, 2017-05-12:
could you please post details from the browser's `Developer Tools / Console` logs?

also run `pm2 logs` after a restart. Then make the error happen. Please post what it starts logging after that point.

Whytehorse, 2017-05-12:
pm2 logs return:
`[TAILING] Tailing last 15 lines for [all] processes (change the value with --lines option)
/home/ben/.pm2/pm2.log last 15 lines:
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:31: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:32: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:32: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:32: pid=7749 msg=failed to kill - retrying in 100ms
PM2        | 2017-05-11 19:34:32: Process with pid 7749 still alive after 1600ms, sending it SIGKILL now...
PM2        | 2017-05-11 19:34:32: App [camera] with id [0] and pid [7749], exited with code [0] via signal [SIGKILL]
PM2        | 2017-05-11 19:34:32: pid=7749 msg=process killed
PM2        | 2017-05-11 19:34:32: Starting execution sequence in -fork mode- for app name:camera id:0
PM2        | 2017-05-11 19:34:32: App name:camera id:0 online

/home/ben/.pm2/logs/camera-error-0.log last 15 lines:
0|camera   |     at process._tickDomainCallback (internal/process/next_tick.js:128:9)
0|camera   | uncaughtException TypeError: tx is not a function
0|camera   |     at Socket.<anonymous> (/home/ben/Shinobi/camera.js:1546:13)
0|camera   |     at emitOne (events.js:96:13)
0|camera   |     at Socket.emit (events.js:191:7)
0|camera   |     at /home/ben/Shinobi/node_modules/socket.io/lib/socket.js:503:12
0|camera   |     at _combinedTickCallback (internal/process/next_tick.js:73:7)
0|camera   |     at process._tickDomainCallback (internal/process/next_tick.js:128:9)
0|camera   | uncaughtException TypeError: tx is not a function
0|camera   |     at Socket.<anonymous> (/home/ben/Shinobi/camera.js:1546:13)
0|camera   |     at emitOne (events.js:96:13)
0|camera   |     at Socket.emit (events.js:191:7)
0|camera   |     at /home/ben/Shinobi/node_modules/socket.io/lib/socket.js:503:12
0|camera   |     at _combinedTickCallback (internal/process/next_tick.js:73:7)
0|camera   |     at process._tickDomainCallback (internal/process/next_tick.js:128:9)

/home/ben/.pm2/logs/camera-out-0.log last 15 lines:
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
0|camera   | Shinobi - PORT : 8080, NODE.JS : v7.10.0
0|camera   | 
`
developer tools console returns:
`ncaught TypeError: Cannot read property 'reDrawMonGroups' of undefined(…)$.ccio.init @ main.dash2.js:100(anonymous function) @ main.dash2.js:523Emitter.emit @ socket.io.js:7414Socket.emit @ socket.io.js:6994Socket.onconnect @ socket.io.js:7188Socket.onpacket @ socket.io.js:7084(anonymous function) @ socket.io.js:7520Emitter.emit @ socket.io.js:7414Manager.ondecoded @ socket.io.js:2823(anonymous function) @ socket.io.js:7520Emitter.emit @ socket.io.js:2285Decoder.add @ socket.io.js:1054Manager.ondata @ socket.io.js:2813(anonymous function) @ socket.io.js:7520Emitter.emit @ socket.io.js:6102Socket.onPacket @ socket.io.js:3523(anonymous function) @ socket.io.js:3341Emitter.emit @ socket.io.js:6102Transport.onPacket @ socket.io.js:4760Transport.onData @ socket.io.js:4752ws.onmessage @ socket.io.js:6640`

moeiscool, 2017-05-12:
i get this error when i try to access the page during an update.

Are you on Chromium? does it break on mobile devices too?
this doesn't require a whole server restart to fix. its a client side thing, you should be able to do a hard refresh.

Its apparently unable to find data about your user or its missing some code because it didn't finish uploading to the server.

Whytehorse, 2017-05-12:
Sometimes a refresh fixes it, sometimes not. Looks like F5 fixes it.

Whytehorse, 2017-05-12:
I can't reproduce this in firefox so it's definitely a chrome thing

moeiscool, 2017-05-12:
hm, i use chrome on windows 7 to test all day. With ad blocker. Is it possible to disable blur and metamask on your shinobi dashboard and leave it on for other sites?

i have a donation button in the help window. I suppose I'll reduce it to just a link if there is a possibility its halting the code somewhere. `/web/pages/blocks/help.ejs`

moeiscool, 2017-05-13:
does this still happen with  blur, metamask, and social fixer disabled?

moeiscool, 2017-05-14:
i believe i have fixed this. I think it was happening because the socket was getting the connection and tried to run the initiator before these variables were set. I have added a `$(document).ready` to it and i haven't had the error since.

closed 2017-05-14 by moeiscool

Whytehorse, 2017-05-14:
What file do I add that to? Can I just re-pull the code? I forgot how to re-pull, lol

moeiscool, 2017-05-14:
yes re-pull, or re download and overwrite `web` and `camera.js`

Whytehorse, 2017-05-14:
Yup that fixed it! Please close. FYI the command is:
`git stash save --keep-index
git pull`