moeiscool/Shinobi #138: UI Not showing viceo recordings older than 17 hours

ToroNZ, 2017-05-12:
I've set to keep recordings for 14 days:
[image]

Files are being saved as expected:
[image]

But when I click on the camera video history-gram oldest shows:
[image]

I've checked for deletions on the camera.js logs and nothing looks to be deleted:
[image]

DB records look fine:
[image]

moeiscool, 2017-05-12:
did this start happening after a recent update?

you have 6 db records but only 2 show in the dashboard? are you seeing records when they first are made and they just gone upon later review? like after you've logged out and logged in?

Could you check your filters?

also that reminds me you can force the cron to rerun and we can know for sure if its deleting stuff.
clear the logs
`pm2 flush`
restart cron and open logs to see what it does.
`pm2 restart cron.js && pm2 logs`

moeiscool, 2017-05-12:
also please try setting a wide date range on the table. I found myself running in circles once forgetting i had the date range thing :sweat_smile:

ToroNZ, 2017-05-12:
No, I've set this up a week ago (haven't touched it since).

I have 1050 DB records for HIK1.
I have 1050 mp4 files for HIK1 in the filesystem.
I've selected a wide range (screenshot shows from 5th of May till 12th of May).

Cron.js restarts every hour (had issues with it not cycling properly). Cron looks to be running and doing it's job.

My suspicion is that the javascript code that queries the DB is getting truncated or not passing the date properly. 

Here's what happens on the UI side, it just won't show anything beyond the past 17 hours:
[image]

ToroNZ, 2017-05-12:
Is there a way to change the date format? i.e. not US format

ToroNZ, 2017-05-12:
Sorry, missed your question:

Issue is persistent across login/logout. Records disappear from the UI after 17 hours, but as per the settings they are not being deleted from the DB or the file-system. So it's pretty much cosmetic.

I've got no filters set for ffmpeg, video-wise all looks and works as expected.

moeiscool, 2017-05-12:
currently there is no way of modifying the format. Can you ensure you are on the latest version of node.js?
`node -v` should give you the version.

When you do an API request for videos, are all videos there? or is the JSON string missing videos.

Can i get remote access to this installation? maybe I can poke around and find what's going on. As it does seem like a cosmetic issue.

moeiscool, 2017-05-13:
closing this since after you updated from 6.10.2 to 7.10.0 it started behaving correctly.

There is a limit of 100 for rows, and I will be adding pagination soon.

closed 2017-05-13 by moeiscool