moeiscool/Shinobi #142: Firefox CSS - Oversized window popup

ToroNZ, 2017-05-12:
Two screenshots which show what I'm experiencing:
Chrome/Epiphany browser:
[image]

Firefox:
[image]

You just cannot see the bottom of the popup window which makes you double scroll to get there.

ToroNZ, 2017-05-12:
Also, Chrome/Epiphany won't show delete button

ToroNZ, 2017-05-12:
Tested on Windows10 and Ubuntu16, consistent behaviour on both platforms.
Something's up with the way Firefox interprets CSS.

moeiscool, 2017-05-12:
ah i see it on my Firefox, windows 7. My chrome looks fine.

ah it looks like firefox is unable to parse the CSS property `max-height: calc(100% - 200px);`

moeiscool, 2017-05-19:
added `height:70%` for firefox only

closed 2017-05-19 by moeiscool