moeiscool/Shinobi #154: API "start" after API "record" results in an unplayable ffmpeg output

pashdown, 2017-05-18:
Cameras that can issue a trigger after motion detection ends can only use "stop" in the API if the recording needs to be retained.  Returning to monitor mode with "start" after a "record" is issued results in a corrupt ffmpeg file.

moeiscool, 2017-05-18:
we talked about this earlier today. I'll post what I had said about it for others to see.
```
unplayable files are sometimes a result of how the camera makes the stream. I have 3 different kinds of cameras. RTSP, JPEG, MJPEG. and some will allow me to cut the video file manually, the video files play. whereas others will simply give you a "broken" file. in most cases this file is not actually broken, its just not "finished". It needs to be processed by ffmpeg one more time mp4 to mp4 or webm to webm, this makes it a working file.

i have yet to add an option to "fix" files.
```

moeiscool, 2017-05-19:
fix function has been added, it may need some tweaking but i think its doing what is expected of it.

closed 2017-05-19 by moeiscool