moeiscool/Shinobi #157: New supported camera types

pschmitt, 2017-05-19:
I've set up Shinobi with a bunch of cameras.

Would be nice to see them included in the [supported camera list](http://shinobi.video/docs/supported)

| Model               | Resolution | Type        | Port | Stream URL      | Save | JSON Template                                                                                                                         | Note                                                          |
|---------------------|------------|-------------|------|-----------------|------|---------------------------------------------------------------------------------------------------------------------------------------|---------------------------------------------------------------|
| Foscam C1           | 1280x720   | H264 (RTSP) | 554  | /videoMain      | WebM | [foscam-c1.json](https://gist.github.com/pschmitt/0b2aae837eeef5a819617eb70267c311/raw/5f49e7d7ceedddb1682dc7e7a8e46c8c04154279/foscam-c1.json)         |                                                               |
| Foscam C2           | 1920x1080  | H264 (RTSP) | 554  | /videoMain      | WebM | [foscam-c2.json](https://gist.github.com/pschmitt/0b2aae837eeef5a819617eb70267c311/raw/5f49e7d7ceedddb1682dc7e7a8e46c8c04154279/foscam-c2.json)         |                                                               |
| mjpg-streamer       | Custom     | MJPEG       | 8080 | /?action=stream | WebM | [mjpg-streamer.json](https://gist.github.com/pschmitt/0b2aae837eeef5a819617eb70267c311/raw/5f49e7d7ceedddb1682dc7e7a8e46c8c04154279/mjpg-streamer.json)     |                                                               |
| IP Webcam (Android) | Custom     | H264 (HTTP) | 8080 | /video          | WebM | [android_ip_webcam.json](https://gist.github.com/pschmitt/0b2aae837eeef5a819617eb70267c311/raw/5f49e7d7ceedddb1682dc7e7a8e46c8c04154279/android_ip_webcam.json) |  https://play.google.com/store/apps/details?id=com.pas.webcam |

moeiscool, 2017-05-19:
sweet! thank you! I am actually in the process of building a list of connection URLs.

I will have a separate listing for templates and ill somehow make the templates appear in the list of cameras as well.

moeiscool, 2017-05-20:
Camera connection URLs can be found here now http://shinobi.video/docs/cameras

Ill probably be redoing the supported page in a few days. I'll add your submissions to the list :) thx!

closed 2017-05-20 by moeiscool