moeiscool/Shinobi #158: API force record

pashdown, 2017-05-22:
Camera motion detectors are really stupid.  They can send a motion detection event in one second and a stop motion the next.  Can you have a command to record for a set period of time, even if start/stop is received, the return to the prior mode before the record was issued?

moeiscool, 2017-05-22:
if you can make it so you don't send a command on stop then yes. The API method to change modes has a timer function also. Check out the docs for more info. http://shinobi.video/docs/api#content-monitor-triggers

Also I like to use the issues section for only issues :) please find me on discord :D

pashdown, 2017-05-22:
Sending a subsequent start 2 seconds after a 30 second record stops the recording and returns it to monitor mode.  I guess this is an issue? :)

moeiscool, 2017-05-22:
Resolved! Our convo is pasted below. The solution was to disable the command to engage the camera with `/start`. Then to add `?reset=1` to the `/record/30` query.

Example : 
```
http://xxx.xxx.xxx.xxx/API_KEY/monitor/2Df5hBE/bunny/record/30/min?reset=1
```

thanks :D I will update the docs to reflect this option.

> pashdown - Today at 2:38 PM
> moe, i should have been more clear on that issue.  if you issue a "start" 2 seconds after issuing a "record/30" to the API, Shinobi stops the recording and returns to monitor mode.

> moeiscool - Today at 2:39 PM
> ah i see what you are saying
> i was a little confused
> well its supposed to be like that , it doesnt queue the request. it overtakes the last one if its different
> like if you do record/30 then record/30 15 seconds later it will be going on for a total of 45 seconds

> pashdown - Today at 2:40 PM
> no, that is fine.  what i need is to override the jittery "motion stopped" detection on the camera.  i want it to record for a minimum of whatever time i send to the API.

> moeiscool - Today at 2:41 PM
> why are you issuing a start command?
> can you stop that one?

> pashdown - Today at 2:42 PM
> hmmm.. good point.  i guess i could.
> ok, just tried sending two records.  it didn't add.  i think i've got the latest version.

> moeiscool - Today at 2:44 PM
> it didnt record anything?

> pashdown - Today at 2:45 PM
> it recorded, but when i sent the second one it, said, ""ok": false,
>    "msg": "Monitor mode is already : record""

> moeiscool - Today at 2:45 PM
> oh woops
> add `?reset=1`
> let me confirm that
> yah that should do it
> you should see "Reset Timer"

closed 2017-05-22 by moeiscool